MrYsLab/telemetrix #1: more complex data types?

ipsod, 2020-09-28:
Hi Mr. Y.  Is there or will there be a way to send commands to the MCU with more types of arguments (int, long, float, char, String, etc.)?  Also, to send such variables (and maybe commands with such arguments) back from the MCU?

MrYsLab, 2020-09-28:
I am concentrating on the types that I need, but there is nothing stopping one from adding code to transfer other types across the serial link. The values would need to be broken down into bytes, since the serial link cannot send anything larger. And of course the value would need to be reassembled when received.

Once the project is pretty much complete, I want to explore using  [ArduinoJson](https://arduinojson.org/) for sending data across the link. Is supports the MessagePack protocol as well as JSON. I have no idea of how much overhead in terms of processing time and memory space this will add, but reading the description on their web page, it sounds like it has miminal impact. If this works out, it should make sending any data type across the link trivial without additional code changes.

I am also interested in exploring the [embedded template library ](https://www.etlcpp.com/). It has several features, such as task scheduling and message routing. 

So to answer your question more directly, I am not intending on adding support for those types, but if it is something you need, hopefully adding the functionality yourself should be a lot more straightforward that it would be for Firmata.

I am in the process in refactoring the sketch into a library that can be installed using the Arduino Library Manager. I am also looking at reorganizing  the code so that all the items one would need to add a new command or report would be found at the top of the file, clearly commented.

ipsod, 2020-09-28:
Thank you for the great answer!  I think I can work with Telemetrix and ArduinoJson.  I'm looking at both right now to learn enough find the right place and way to add `send_messagepack` and `receive_messagepack` functions to each end.

MrYsLab, 2020-09-28:
Great! Let me know if you are successful and also any problems you might encounter along the way.

MrYsLab, 2020-10-13:
I have benchmarked  ArduinJson and documented the results on [my blog.](https://mryslab.github.io/bots-in-pieces/arduino,telemetrix,arduinojson/2020/10/10/telemtrix-arduinojson.html) The performance is very bad and therefore, it is not really useful for Telemetrix. So if you need to send types larger than a byte across the link, you could create the disassembly and assembly functions and methods and add that to the code. In the current code, the only data type larger than a byte that I am sending across the link is an integer. To conserve space on the Arduino, I will not be providing conversion functions. Here are links on how I am currently converting an integer to bytes [(Arduino)](https://github.com/MrYsLab/telemetrix/blob/f9945a380a15f52fa98ee8c72011aa2ab97befd7/phase4/Telemetrix4Arduino/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L263) [(Python)](https://github.com/MrYsLab/telemetrix/blob/f9945a380a15f52fa98ee8c72011aa2ab97befd7/phase4/telemetrix/telemetrix.py#L608)

And bytes to integer ([Arduino](https://github.com/MrYsLab/telemetrix/blob/f9945a380a15f52fa98ee8c72011aa2ab97befd7/phase4/Telemetrix4Arduino/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L618)) and [(Python)](https://github.com/MrYsLab/telemetrix/blob/f9945a380a15f52fa98ee8c72011aa2ab97befd7/phase4/telemetrix/telemetrix.py#L738).

I am closing this issue, but if you have any comments, if you post them here I will see them.

Thanks for the suggestion.

closed 2020-10-13 by MrYsLab