MrYsLab/telemetrix #3: Allow for multiple I2C interfaces

mklomp, 2020-10-07:
This issue is already in [Firmata for a long time](https://github.com/firmata/arduino/issues/214). But since you are redesigning the whole thing anyway, it would be a really nice feature to have this in telemetrix. At least the STM32, but also the Due have multiple I2C interfaces.

MrYsLab, 2020-10-07:
The i2c pins are defined and supported within the STM32DUINO Arduino core. Currently there is only one i2c interface defined.
[Issue #705](https://github.com/stm32duino/Arduino_Core_STM32/issues/705) against the STM32DUINO Arduino Core requests this feature also. When they implement, I will support it.

MrYsLab, 2020-10-21:
Release 0.1.7 implements this feature

closed 2020-10-21 by MrYsLab