MrYsLab/telemetrix #6: If com_port is specified for a manal connect, board is not recognized.

MrYsLab, 2020-11-11:
The threads were never started in this scenario. Need to start the threads

MrYsLab, 2020-11-15:
Resolved with release 1.3

Kaladin-Syl-WR, 2021-02-01:
Good day. I realize that this issue is closed, but I have Telemtrix v1.5 installed, and I do have this issue with the manual com_port spesification. When I don't manually initialize the com_port, it works fine, but as soon as I specify a port manually, I get the following error:

[image]

Thank you in advance.

MrYsLab, 2021-02-01:
How are you instantiating the Telemetrix class? I just tried using the blink example both as-is (auto-discovery of the port ) and manually specifying it by changing the instantiation line to 
```
# Create a Telemetrix instance.
board = telemetrix.Telemetrix(com_port='/dev/ttyACM0')

```
Both cases work for me.
Do you have the com port string in quotes?

Kaladin-Syl-WR, 2021-02-02:
Well I put it in double quotes. This is how I instantiated a Telemtrix class:

```
board = telemetrix.Telemetrix(com_port="COM7")
```

MrYsLab, 2021-02-02:
@Kaladin-Syl-WR 
Just to make sure that this is not a Windows issue, I tested it on my Windows box and it works.  To try and debug your issue please try the following steps:

1.Plug your board into your USB port and then open a command window and type:
```
python -m serial.tools.list_ports
```
This will list all of the serial ports in use. For me, there is only one, com5.
[image]

If when you run this utility there is more than one com port listed, make sure you are selecting the correct one.

2. Now using the com port you wish to test, edit the following program for your specific com port. You don't have to connect an LED to the board. Just run the program. If you name it blink2.py then run it as:
```
python blink2.py
```

```

import sys
import time

from telemetrix import telemetrix

"""
Setup a pin for digital output 
and toggle the pin 5 times.
"""

# some globals
DIGITAL_PIN = 6  # the board LED

# Create a Telemetrix instance.
board = telemetrix.Telemetrix(com_port='com5')

# Set the DIGITAL_PIN as an output pin
board.set_pin_mode_digital_output(DIGITAL_PIN)

# Blink the LED and provide feedback as
# to the LED state on the console.
for blink in range(5):
    # When hitting control-c to end the program
    # in this loop, we are likely to get a KeyboardInterrupt
    # exception. Catch the exception and exit gracefully.
    try:
        print('1')
        board.digital_write(DIGITAL_PIN, 1)
        time.sleep(1)
        print('0')
        board.digital_write(DIGITAL_PIN, 0)
        time.sleep(1)
    except KeyboardInterrupt:
        board.shutdown()
        sys.exit(0)
```
Please let me know if this program runs. If you still are having issues, please post your Python code so I can look at it.

MrYsLab, 2021-02-04:
I am closing this issue since I am unable to reproduce what you are seeing. If you are still experiencing the issue, please provide a Python file so that I may try to reproduce the issue here. If necessary I will reopen the issue.

closed 2021-02-04 by MrYsLab

Kaladin-Syl-WR, 2021-02-08:
Sorry for only answering now. Everything is working fine now. Not quite sure what was wrong, as it was not the com port.

None the less, thank you for your effort and help.