MrYsLab/telemetrix #14: wrong arduino id

builderjer, 2021-06-20:
The sketch seems to install to the arduino fine, but when i initiate ```t = telemetrix.Telemetrix()``` it goes through the process and says it is the wrong arduino id.  This is with the examples and my own program.

MrYsLab, 2021-06-20:
I just ran the blink script on an Arduino Leonardo without issue, and here is the output:
```
Telemetrix:  Version 1.6

Copyright (c) 2020 Alan Yorinks All Rights Reserved.

Opening all potential serial ports...
	/dev/ttyACM0

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...
Arduino compatible device found and connected to /dev/ttyACM0
Waiting for Arduino to reset
Reset Complete
Retrieving Arduino ID...
Valid Arduino ID Found.

Retrieving Telemetrix4Arduino firmware ID...
Telemetrix4Arduino firmware version: 2.0.0
1
0
1
0
1
0
1
0
1
0
```

I apologize, but I will need to ask several questions to try and get to the bottom of your issue.
1. What board are you running? (Arduino Uno, Leonardo, etc.)?
2. Do you have version 2.0.0 loaded on the board? This is the sketch loaded on the board.
3. Are you using version 1.6 of Telemetrix? This is the python client.
4. What operating system does your PC use -Windows, Linux, macOS. If you are using a Raspberry Pi, are you running the latest version of Raspberry Pi OS?
5. Could you please run the following line in a terminal window:
```
python3 -m serial.tools.list_ports
```
Is the port that your board is connect to shown and does telemetrix pick that port to connect to?

6. Could you please try the following: Using the blink.py script, power up your board and wait about 15 seconds before starting the blink.py script. Does this work?

builderjer, 2021-06-20:
Thank you for your quick reply.
I'll try and answer your questions

1. Arduino Uno(Generic)
2. Yes, ```git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```
3. Yes, ```pip show telemetrix
Name: telemetrix
Version: 1.6
Summary: Remotely Control And Monitor Arduino-Core devices
Home-page: https://github.com/MrYsLab/telemetrix
Author: Alan Yorinks
Author-email: MisterYsLab@gmail.com
License: UNKNOWN
Location: /home/jbrodie/Software/brodie_house/venv/lib/python3.9/site-packages
Requires: pyserial
Required-by: ```
4. I run arch linux on relatively new hardware
5.  ```python3 -m serial.tools.list_ports
/dev/ttyACM0        
1 ports found
```
6. ```python blink.py 
Telemetrix:  Version 1.6

Copyright (c) 2020 Alan Yorinks All Rights Reserved.

Opening all potential serial ports...
	/dev/ttyACM0

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...
Arduino compatible device found and connected to /dev/ttyACM0
Waiting for Arduino to reset
Reset Complete
Retrieving Arduino ID...
Traceback (most recent call last):
  File "/home/jbrodie/Software/brodie_house/arduino/telemetrix/examples/blink.py", line 35, in <module>
    board = telemetrix.Telemetrix()
  File "/home/jbrodie/Software/brodie_house/venv/lib/python3.9/site-packages/telemetrix/telemetrix.py", line 240, in __init__
    raise RuntimeError(f'Incorrect Arduino ID: {self.reported_arduino_id}')
RuntimeError: Incorrect Arduino ID: []
```

I hope this helps.  Funny thing is, telemetrix-aio works fine

MrYsLab, 2021-06-20:
This is strange behavior that I have not seen before. I don't think Arch has anything to do with it, but I will load arch  2021.06.01 into Virtualbox. I will also run on an Uno in case there is some timing issue sitting on the edge.
I will get back with the results.

MrYsLab, 2021-06-20:
I could not get Arch installed in VirtualBox and working properly.
However, I created a [branch ](https://github.com/MrYsLab/telemetrix/tree/long_id_wait)that lengthens the time to receive the Arduino ID. 
Do you feel confident enough to install either using a local pip install or using setup.py?
The version in this branch was bumped up to 1.7 to identify that you are running the updated version.
In either case, [download the .zip ](https://github.com/MrYsLab/telemetrix/archive/refs/heads/master.zip)
Rename telemetrix-long_id_wait to telemetrix and to install with setup.py, cd into telemetrix and type:
```
sudo python3 setup.py
```
For a local pip install, do not cd into telemetrix, but stay at the directory above and type:
```
sudo pip3 install ./telemetrix --upgrade
```
Please let me know if this helps anything?
If this works, I will make an official release.
Thanks.

builderjer, 2021-06-20:
```
python blink.py 
Telemetrix:  Version 1.7

Copyright (c) 2020 Alan Yorinks All Rights Reserved.

Opening all potential serial ports...
	/dev/ttyACM0

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...
Arduino compatible device found and connected to /dev/ttyACM0
Waiting for Arduino to reset
Reset Complete
Retrieving Arduino ID...
Traceback (most recent call last):
  File "/home/jbrodie/Software/brodie_house/arduino/telemetrix/examples/blink.py", line 35, in <module>
    board = telemetrix.Telemetrix()
  File "/home/jbrodie/Software/brodie_house/venv/lib/python3.9/site-packages/telemetrix/telemetrix.py", line 240, in __init__
    raise RuntimeError(f'Incorrect Arduino ID: {self.reported_arduino_id}')
RuntimeError: Incorrect Arduino ID: []
```

I'm not sure why this doesn't work, but telemetrix-aio does

MrYsLab, 2021-06-20:
I am at a loss to understand what might be the problem. Out of curiosity, what version of Python are you running? I will spend some more time trying to get a working version of Arch installed, but I have the feeling that will probably lead me to nowhere.

builderjer, 2021-06-20:
Ya me too, especially since the aio version works.

Python 3.9.4 in a venv

MrYsLab, 2021-06-20:
Thanks. I got networking working in Arch, but now I can sudo anything. I will reinstall and give my login root privileges. Also, will try 3.9.4 in Ubuntu.

MrYsLab, 2021-06-20:
I finally got Arch up and running. I ran blink.py in a venv and did not have any issues. I loaded the latest version of Arch, and it has python 3.9.5. At this point, I am not sure what to look at. Do you run any virtual environments on your computer, such as VirtualBox or VMWare? If so, perhaps try installing another distro such as Ubuntu as a virtual machine and see if you are seeing the same thing. I will keep thinking about.

builderjer, 2021-06-20:
Maybe I will try 3.9.5 and see if that is the issue.

builderjer, 2021-06-20:
fresh venv of 3.9.5 and still the same thing.  I'm going to dig up another arduino and give it a shot

MrYsLab, 2021-06-22:
I restored the wait time to the original value in the long_id branch. I then added some debug prints. Here is the output I see when I run the blink script. If you run this branch, do you see something similar?
```
starting the serial receiver thread
starting reporter thread
starting data_receiver thread
Telemetrix:  Version 1.7

Copyright (c) 2020 Alan Yorinks All Rights Reserved.

Opening all potential serial ports...
	/dev/ttyACM0

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...
Arduino compatible device found and connected to /dev/ttyACM0
Waiting for Arduino to reset
Reset Complete
Retrieving Arduino ID...
serial received started. shutdown flag = False
reporter thread started
report type= 6
Valid Arduino ID Found.

Retrieving Telemetrix4Arduino firmware ID...
report type= 5
Telemetrix4Arduino firmware version: 2.0.0
1
0
1
0
1
0
1
```
This is telling me that all the threads have started successfully and are running.
The report type of 6 is the valid Arduino ID report, and 5 is the firmware version report.

If you run this, do the threads seem to be up and running? If you get the same error as before, the last message you will probably see is "reporter thread started'.

Another thing you might try is to increase the arduino_wait parameter when instantiating telemetrix.

MrYsLab, 2021-06-26:
@builderjer 
Have you been able to resolve the issue?

builderjer, 2021-06-26:
I'm sorry, life has gotten in the way and I have not been able to try again. As soon as I do, I will let you know.

MrYsLab, 2021-06-27:
Thanks. No rush.

builderjer, 2021-07-03:
finally had a chance to try this.
These are my steps
```
pip install git+https://github.com/MrYsLab/telemetrix.git@long_id_wait
Collecting git+https://github.com/MrYsLab/telemetrix.git@long_id_wait
  Cloning https://github.com/MrYsLab/telemetrix.git (to revision long_id_wait) to /tmp/pip-req-build-bpjp9k0x
  Running command git clone -q https://github.com/MrYsLab/telemetrix.git /tmp/pip-req-build-bpjp9k0x
  Running command git checkout -b long_id_wait --track origin/long_id_wait
  Switched to a new branch 'long_id_wait'
  Branch 'long_id_wait' set up to track remote branch 'long_id_wait' from 'origin'.
Requirement already satisfied: pyserial in ./venv/lib/python3.9/site-packages (from telemetrix==1.7) (3.5)
Building wheels for collected packages: telemetrix
  Building wheel for telemetrix (setup.py) ... done
  Created wheel for telemetrix: filename=telemetrix-1.7-py3-none-any.whl size=12791 sha256=6e5a6627db881f6769f0f317cf4573164fc11f554d2dfd1937c333646e4d82e2
  Stored in directory: /tmp/pip-ephem-wheel-cache-71o3v_la/wheels/ee/22/96/5461344f2ef041cd185320eb2dbaa78dc138969037b3df5ed0
Successfully built telemetrix
Installing collected packages: telemetrix
Successfully installed telemetrix-1.7
```

And then tried the blink.py

```
python arduino/telemetrix/examples/blink.py 
starting the serial receiver thread
starting reporter thread
starting data_receiver thread
Telemetrix:  Version 1.7

Copyright (c) 2020 Alan Yorinks All Rights Reserved.

Opening all potential serial ports...
	/dev/ttyACM0

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...
Arduino compatible device found and connected to /dev/ttyACM0
Waiting for Arduino to reset
Reset Complete
Retrieving Arduino ID...
reporter thread started
serial received started. shutdown flag = False
Traceback (most recent call last):
  File "/home/jbrodie/Software/brodie_house/arduino/telemetrix/examples/blink.py", line 35, in <module>
    board = telemetrix.Telemetrix()
  File "/home/jbrodie/Software/brodie_house/venv/lib/python3.9/site-packages/telemetrix/telemetrix.py", line 243, in __init__
    raise RuntimeError(f'Incorrect Arduino ID: {self.reported_arduino_id}')
RuntimeError: Incorrect Arduino ID: []
```

I don't know what the issue is. I will try another computer.

MrYsLab, 2021-07-03:
This is bizarre behavior, especially since you mentioned that it works with the aio version. There is definitely a communication error. Either the sketch never sees the request for the ID, or the Python side never sees the reply.  One thing you can try is to run the following:
```
import time
from telemetrix import telemetrix

board = telemetrix.Telemetrix()
time.sleep(5)
```
When you run this, look at the Rx and Tx LEDs. They both should flash. 

The output you should see is:
```
Telemetrix:  Version 1.6

Copyright (c) 2020 Alan Yorinks All Rights Reserved.

Opening all potential serial ports...
	/dev/ttyACM1

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...
Arduino compatible device found and connected to /dev/ttyACM1
Waiting for Arduino to reset
Reset Complete
Retrieving Arduino ID...
Valid Arduino ID Found.

Retrieving Telemetrix4Arduino firmware ID...
Telemetrix4Arduino firmware version: 2.0.0
```
Have you tried a different USB cable?

builderjer, 2021-07-04:
Tried that, tried a different usb, tried a different computer running arch, even tried a different arduino.  The only thing that I can figure is maybe a permission issue I have.  The odd thing is that the AIO version works fine.  Your arch test was a virtual machine, so maybe that is the difference also.  I just don't know.  Thanks for all your help

MrYsLab, 2021-07-04:
Very confusing. If you run the program above, modified to lengthen the Arduino wait time, do you see the RX and TX LEDs flash after the 10 second wait time?
```
import time
from telemetrix import telemetrix

board = telemetrix.Telemetrix(arduino_wait=10)
time.sleep(5)
```
If they flash and you are still getting the exception, I do not know what it might be.

builderjer, 2021-07-04:
just the green one flashes

MrYsLab, 2021-07-04:
Below is a photo of an Uno. The RX and TX LEDs are within the red rectangle. I am assuming that the LED that is blinking on your board is the RX LED. If that is the case, the sketch receives the command, but does not reply. If neither blink, then the command is never being seen by the Arduino.
Out of curiosity, is your board an "official" Arduino board or a clone?  I am testing with both an official board and a clone, and both work here.

[image]

builderjer, 2021-07-04:
I have a couple of different off brands of the uno, and one of them is not laid out the same as that.  Either way, both versions only flash the rx quickly and then the error

MrYsLab, 2021-07-04:
So, for some reason, the sketch is not replying. To look at what the sketch is receiving,  you can uncomment out[ this line ](https://github.com/MrYsLab/Telemetrix4Arduino/blob/001470552bbc7de6abd8ffd724dde65279cbb4ad/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L745)in the sketch and do a run, what is the output to your console?

builderjer, 2021-07-04:
Well I fixed it.  I was using the arduino sketch located at 
telemetrix/arduino_sketches/arduino-telemetrix/arduino-telemetrix.ino

the one you said to modify is located at 
Telemetrix4Arduino/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino

As soon as I installed that version, everything worked correctly.

Mind you, the issue was made on the first repository, not the second.  Thank you for all your time and work

MrYsLab, 2021-07-04:
Thanks. My hope was that folks would use the [User's Guide](https://mryslab.github.io/telemetrix/) for install instructions. I will place a note in the readme that will hopefully save others from what you encountered.
Thanks for your patience.

closed 2021-07-04 by MrYsLab

builderjer, 2021-07-04:
I started with the user's guide, and the first link to the python library has an arduino sketch that I used.
I have been using PyMata4 and saw that most of the commands are similar so I just went for it.  I should have read a little farther.

MrYsLab, 2021-07-04:
No problem. I probably should have had a separate production release repository from the development one. I am putting a readme in the Arduino directory that will hopefully save others from being confused. 
I am glad that all is resolved.