MrYsLab/telemetrix #16: dht_read() doesn't exist.

dlefur, 2021-11-10:
Hello.
I am trying to convert my Pymata examples to telemetrix.
I didn't find any dht_read() method.

Is it normal ? Using a callback seems to be compulsory now. 

Cordially.

MrYsLab, 2021-11-10:
Yes, callbacks are required. The reasoning behind this is that polling for values can slow things down. Also, with polling, your application might miss an event change. With callbacks, an application is notified almost immediately. 
With polling, data locking is employed to ensure that the data is valid when the read is performed.  Data locks slow processing down and add complexity to the code.

dlefur, 2021-11-11:
Thank you for this answer.
When the user wants **only one** measure, I was wondering if using **enable_digital_reporting** and **disable_digital_reporting** will be a solution to activate and disactivate the callback.

By the way, I have noticed that the **play_tone** methods are not in telemetrix. For my young students, it is nice to play with it. I think I will continue using Pymata4.

Finally. Is there a forum ? This was not really an issue. Sorry ...

Cordially.

MrYsLab, 2021-11-11:
Disabling digital and analog reporting will prevent any further notifications for digital and analog inputs. The callback function can also be written to process only the first notification but ignore all future reports.

Play_Tone was removed from Telemetrix because the tone library has issues working with boards other than Arduino boards, specifically for STM-32 boards and the ESP8266. The Pymata clients originally included tone in support of my Scratch extensions. The [latest Scratch3 extension](https://mryslab.github.io/s3-extend/) for Arduino still provides support for the Tone library. Still, since Scratch3 does a much better job producing tones than the Tone library and a piezo buzzer, I am unsure if it is still essential.

If pymata4 meets your needs, by all means, please continue to use it.

I have a couple of Gitter forums set up, but they have had very little activity recently. I have a couple of blogs that have documented some of my "adventures. Some of my older articles can be found [here](http://mryslab.blogspot.com/). The blog that has the most current articles may be found [here](https://mryslab.github.io/bots-in-pieces/index.html). 

If you ever have any questions, you can always send me an email. My email address [can be found underneath the avatar](https://github.com/MrYsLab) on the left side of the page.

closed 2021-11-11 by MrYsLab