MrYsLab/telemetrix #19: stepper_move not working properly with negative numbe

bloodsuckant, 2022-01-12:
Hi, thanks for this beautiful project. I use Telemetrix to control a stepper motor. The stepper_move_to is nice, it working correctly. However, the stepper_move with negative number is not simply counter clockwise from the current position, it will go counter clockwise with extra 65536 steps, i.e. -(65536 + distance_to_go) steps. It is nice if u can fix this. For now, people can use stepper_move_to (absolute) to control the motor nicely.

MrYsLab, 2022-01-12:
Thanks for reporting this. I am able to reproduce the issue and will be working on a fix.

MrYsLab, 2022-01-13:
I just wanted to give you an update. I am working on the fixes, but it will take several days to complete. 

I've decided to convert the client's use of Python's int.to_bytes() to using struct.pack(). The reasoning behind this change is that this will give slightly better performance as [documented in the article](https://www.delftstack.com/howto/python/how-to-convert-int-to-bytes-in-python-2-and-python-3/#performance-comparison). Changes are required both in the client and server.

In addition, since the Telemetrix4Esp8266 server uses the telemetrix and telemetrix-aio clients, I will need to update that as well.

After all the changes, I will need to test all of the iterations of the clients and servers. Finally, when all this is complete, I can publish the changes.

Thanks for your patience.