MrYsLab/telemetrix #32: Documentation error in set_pin_mode_digital_input(_pullup)

Xevel, 2022-06-04:
Hi (again),

Seems to me that this time I have found a real bug!
https://github.com/MrYsLab/telemetrix/blob/bd342a083ee8c263d3f65c8897e5a2e50530dcbe/telemetrix/telemetrix.py#L778
https://github.com/MrYsLab/telemetrix/blob/bd342a083ee8c263d3f65c8897e5a2e50530dcbe/telemetrix/telemetrix.py#L796

The documentation of set_pin_mode_digital_input and set_pin_mode_digital_input_pullup mention a "pin_type" that is supposedly 0 for INPUT (AT_INPUT = 0, that matches) and 11 for INPUT_PULLUP (actually AT_INPUT_PULLUP = 2), but the code generating the list sent to the callback uses instead DIGITAL_REPORT (= DIGITAL_WRITE = 2) for both.

https://github.com/MrYsLab/telemetrix/blob/bd342a083ee8c263d3f65c8897e5a2e50530dcbe/telemetrix/telemetrix.py#L2173

Cheers,
Nicolas

MrYsLab, 2022-06-04:
Actually, this is not a bug. Both _set_digital_input_ and _set_digital_input_pullup_ call a common lower-level method, __set_pin_mode_. The pin mode value is set to the corresponding pin mode defined in Arduino.h. Digital Input is 0, and Digital Input Pullup is 2. AT_INPUT and AT_INPUT_PULLUP are defined in the private_constants.py file. The Telemetrix4Arduino sketch takes the mode passed to it in the message and uses it to call the Arduino method  pin_mode().

In Telemetrix4Arduino, when it scans for digital input changes,[ it treats both input types without distinction](https://github.com/MrYsLab/Telemetrix4Arduino/blob/fa7d30c9f42ccc3953a3e705df5d84ee29b04004/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1740). That is, [the report type is DIGITAL_REPORT for both input types](https://github.com/MrYsLab/Telemetrix4Arduino/blob/fa7d30c9f42ccc3953a3e705df5d84ee29b04004/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1736). 

Using a single report type simplifies things on the Python side. 

I [added notes in the User's Guide ](https://mryslab.github.io/telemetrix/pin_modes/)to help clarify this.

I am closing the issue, but if you feel it needs to be kept open, please let me know, and I will re-open it.

closed 2022-06-04 by MrYsLab

Xevel, 2022-06-04:
My bad, I did not explain myself well.
The bug to me is in the comment that documents the aforementioned functions, not the functional code.

https://github.com/MrYsLab/telemetrix/blob/bd342a083ee8c263d3f65c8897e5a2e50530dcbe/telemetrix/telemetrix.py#L796

If I set up pin 53 as input with pullup, that text makes me think the callback data should look like [11, 53, state, timestamp] when in fact the first item will be 2 (DIGITAL_REPORT)

https://github.com/MrYsLab/telemetrix/blob/bd342a083ee8c263d3f65c8897e5a2e50530dcbe/telemetrix/telemetrix.py#L778

Same without pullup, where the documenting comment would make me think the callback should expect [0, 53, state, timestamp].

Of course your notes in the web documentation are clear and correct, though if the documenting comments were correct too there would not be a need for them - and in any case the comments should be consistent with what is going on in the code.

MrYsLab, 2022-06-04:
You are correct. I just updated the comments in the code and updated the API. The code is available in the repository but won't be part of a release until next. However, the online API has the corrected comments. 

Thanks for pointing this out.

Xevel, 2022-06-06:
Thanks.
https://mryslab.github.io/telemetrix/pin_modes/ still has copies of the erroneous comments.
And I guess the comments of the other modes might need checking too since set_pin_mode_analog_input has a comment saying:
```The pin_type for analog input pins = 2```
Even though 2 is DIGITAL_REPORT.

MrYsLab, 2022-06-06:
It was changed for [set_pin_mode_analog_input ](https://htmlpreview.github.io/?https://github.com/MrYsLab/telemetrix/blob/master/html/telemetrix/index.html#telemetrix.telemetrix.Telemetrix.set_pin_mode_analog_input)as well.

I have not created a new release with the comment changes, so if you are grabbing the release .zip or tar.gz, these won't contain the changes. However, if you download from the repository with the [Code](https://github.com/MrYsLab/telemetrix/archive/refs/heads/master.zip) link, the changes will be present in the files.

Sorry about the confusion.