MrYsLab/telemetrix #38: Add support for FastAccelStepper?

ipsod, 2022-08-05:
Hello!

[FastAccelStepper](https://github.com/gin66/FastAccelStepper) seems to be a significant upgrade from AccelStepper.  I plan to try to integrate it myself in the future, but hoped, if you're interested, it might be finished before I got around to starting.

MrYsLab, 2022-08-05:
@ipsod Thanks for the suggestion. I have no plans to implement this for the following reasons:
1. The pins it supports for the step signal are extremely limited.
2. It does not support all of the boards that the Teletetrix clients and servers support. For example, the esp8266.

If you wish to modify the client and server for your board, I would be happy to answer any questions you might have but will not accept a pull request to add support for FastAccelStepper. 

Since I have no plans to expand the functionality of Telemetrix at this time, creating a private branch of the code would be a very low risk for you.

I am closing this issue, but I will still see the comments if you wish to comment.

Thanks again for your suggestion.

closed 2022-08-05 by MrYsLab

ipsod, 2022-08-05:
Ok, thank you for the quick response.  I'm a big fan of your work.