MrYsLab/telemetrix #49: Logging module compatibility

dirtbirb, 2022-11-16:
Is there a way to manage console output from Telemetrix using the standard Python logging module? Maybe just by turning `print`s into `logging.debug` or whatever level is appropriate?

I'm using a Kivy app to control various devices, some of which run through a Telemetrix-controlled Arduino. Kivy has this nicely-formatted variation on the standard Python logger, and currently my non-Telemetrix devices can tell me what's going on with statements like `logging.warn("Thing: had a problem!")`, which will display normally when running by themselves or matching Kivy's format when run by the GUI. It'd be really nice to manage Telemetrix's output using this same system to keep my console output readable.

I'd submit a pull request with every Telemetrix `print` replaced by a `logging.debug` (or whichever level seems appropriate), but frankly I'm afraid to touch things without asking somebody who knows what they're doing. I'm also not sure if there's just already a switch somewhere that achieves the same effect.

Thanks!

MrYsLab, 2022-11-17:
@dirtbirb Thanks for your suggestion, but logging won't be added to telemetrix at this time. Currently, there are 11 "flavors" of Telemetrix clients, and since I strive for consistency between them, I do not have the time to write and test that feature. 
Please feel free to fork the repository and make any changes you wish as long as you stay within the confines of the AGPL license.
If you have any questions, please feel free to contact me. I am closing this issue, but if you may still post a reply and I will see it.
Thanks.

closed 2022-11-17 by MrYsLab

dirtbirb, 2022-11-17:
Understood, thanks for the reply!