MrYsLab/telemetrix #52: High CPU usage?

chconnor, 2022-11-27:
Hello again -- with this demo code:

```from telemetrix import telemetrix
import time, sys

try:
    mcu = telemetrix.Telemetrix()
    while True: time.sleep(.5)
except KeyboardInterrupt:
    mcu.shutdown()
    sys.exit(0)
```
...I'm seeing 60-80% CPU usage for the python process (using top). It's not a fast computer ([passmark 1992](https://www.cpubenchmark.net/cpu.php?cpu=Intel+Celeron+N4500+%40+1.10GHz&id=4227)) but still... seems like it shouldn't be using hardly any CPU, even if it was reporting pin changes? (If I don't instantiate the Telemetrix object, there is little/no CPU usage.)

My project will need all the available horsepower of the CPU, so I'm crossing my fingers that this is a bug. :-)

chconnor, 2022-11-27:
Aha -- sleep_tune defaults to .0000001. If I set it to something like .1, much better.

I understand that reporting resolution will be coarser, but do I risk a pileup of pin changes bogging down the MCU if Telemetrix doesn't read the serial data fast enough? (Even setting sleep_tune to .01 causes a few percent of constant CPU usage and I'd love to avoid it.)

Any other gotchas to know about with sleep_tune=.1 ?

MrYsLab, 2022-11-27:
First, perhaps this was a cut-and-paste error, but the code you supplied does not run, so I corrected it and made sure it was PEP8 compliant:
```
import sys
import time
from telemetrix import telemetrix

mcu = None
try:
    mcu = telemetrix.Telemetrix(sleep_tune=0.1)
    # mcu = telemetrix.Telemetrix(sleep_tune=0.000001)
    while True:
        try:
            time.sleep(.5)
        except KeyboardInterrupt:
            mcu.shutdown()
            sys.exit(0)
except KeyboardInterrupt:
    mcu.shutdown()
    sys.exit(0)
```

I ran the code on my five-year-old computer, and here are screenshots of the results.

sleep_tune = 0.000001:

[image]

sleep_tune = 0.1 
The screenshot shows zero, but it went up to 0.2 %

[image]

I don't know what is going on with your system, but the numbers you report are suspect.

MrYsLab, 2022-11-27:
I just [looked up](https://www.notebookcheck.net/Intel-Celeron-N4500-Processor-Benchmarks-and-Specs.514524.0.html) your processor, and here is the description:

> Performance
> 
> The average N4500 in our database shapes up to be a very slow processor, its multi-thread benchmark scores only just matching those of [Intel Core i3-4010U](https://www.notebookcheck.net/Intel-Core-i3-4010U-Notebook-Processor.93567.0.html). (The latter saw the light of day in 2013 as a lower mid-range, dual-core CPU designed for use in ultraportable laptops.) In other words, expect the Celeron to be painfully slow in all but the most basic activities such as writing work-related e-mails.

chconnor, 2022-11-27:
Oops, it was a github formatting mistake -- code fixed.

So is the only consequence of a larger sleep_tune value that latency goes up? If I'm also reporting a couple analog pins it's not going to overflow a buffer somewhere, or drop data if sleep_tune == .1, right?

-------------

Yeah, the N4500 is slow; but it did come in a new [MSI Cubi](https://www.newegg.com/p/N82E16883152975?Description=MSI%20Cubi%20N&cm_re=MSI_Cubi%20N-_-83-152-975-_-Product&quicklink=true) that I'm hoping to use for an installed exhibit, hence the exploration here.

What CPU were you testing on? Even on my desktop [i7-4770K](https://www.cpubenchmark.net/cpu.php?cpu=Intel+Core+i7-4770K+%40+3.50GHz&id=1919) (which is old now, too, but not so bad) that demo code in the first post takes 16% of the CPU. I've been trying to figure this out... not sure if any of this is interesting or useful, but just in case:

Seems like maybe you are testing on Windows? It seems like on Linux there are some nonlinear responses in the underlying libraries, maybe having to do with the implementation of the various involved modules, thread switching, etc.

I made the following test code to see if I could reproduce the issue in a stripped-down way. This constructs the two threads like you do in telemetrix.py, and does the same basic operations on every iteration of the sleep_tune loops (with the exception of checking a boolean). It waits 10 seconds, adjusts sleep_tune, and repeats. It wants the arduino attached, obviously.

I'd be curious to know what you see for this test, especially for the smallest sleep_tune value in the test.

For me it uses a bit more CPU than the actual telemetrix code (20% instead of 16%) but it's close enough that maybe it's meaningful. Anyway it doesn't look like there's anything obvious to do about this ("this" being the way that linux requires larger sleep_tune values to be performant on slower CPUs) but I thought it was interesting.

```
import threading, time, serial
from collections import deque

sleep_tune = .000001

run_event = threading.Event()
run_event.set()
# I used COM3 on windows
serial_port = serial.Serial('/dev/ttyACM0', 115200,
                            timeout=1, writeTimeout=0)
serial_port.reset_input_buffer()
serial_port.reset_output_buffer()
dq = deque()

def _reporter():
    while True:
        if len(dq): pass
        if run_event.is_set(): pass
        if serial_port.inWaiting(): pass
        time.sleep(sleep_tune)

def _serial_receiver():
    while True:
        if run_event.is_set(): pass
        if serial_port.inWaiting(): pass
        time.sleep(sleep_tune)

the_reporter_thread = threading.Thread(target=_reporter)
the_reporter_thread.daemon = True
the_data_receive_thread = threading.Thread(target=_serial_receiver)
the_data_receive_thread.daemon = True
the_reporter_thread.start()
the_data_receive_thread.start()

for i in range(3):
    print(f'sleep tune is now {sleep_tune:.6f}')
    time.sleep(10)
    sleep_tune*=1000

```

On Linux in 3.10.6, i7-4770K I get the following CPU usages:
0.000001: 21%
0.001000: 4%
1.000000: ~0%

On Windows in 3.11, i7-4770K I get the following:
0.000001: 24.5%
0.001000: 0.1%
1.000000: ~0%

On Linux in 3.10.6, Celeron N4500:
0.000001: 64%
0.001000: 10%
1.000000: 0.3%

To my eye, the Windows result makes more sense; we are doing 1000X less work and it shows roughly 1000X less CPU usage, but the linux result is only ~5X less from the .000001 to the .001 (on both the i7-4770K and the N4500). However my Windows and Linux values were comparable at .0000001, with Linux being a bit faster.

In fiddling with this, out of 20% usage on linux it seemed like the sleeping itself added like 5%, the use of two threads added about 5%, the serial.inWaiting() added like 10%, and the len(dq) and is_set() didn't add much.

([I did find this thread about inWaiting efficiency](https://stackoverflow.com/questions/68946694/pyserial-in-waiting-cpu-usage)).

MrYsLab, 2022-11-28:
I just ran your original code that I modified on an ancient laptop with [this processor](https://www.cpubenchmark.net/cpu.php?cpu=Intel+Pentium+T2080+%40+1.73GHz&id=586). The results for sleep_tune=0.000001 is between 30% and 37%, and for sleep_tune=0.1, it runs between 1% and 3%.

I ran the tests on a 32-bit version of Linux MX. The laptop is a Toshiba Satellite A135. Python used is version 3.7.3.

So I have no idea why your results are so poor. 

You probably will not have any data loss due to the slower sleep_tune, but I cannot guarantee that. The messages from the Arduino are fairly short, so the UART buffer on the PC will most likely not get overwritten. It all depends upon if the analog inputs are changing rapidly.

chconnor, 2022-11-28:
Weird, thanks for trying it. I just ran it on a [passmark 582 CPU](https://www.cpubenchmark.net/cpu.php?cpu=Intel+Celeron+N2840+%40+2.16GHz&id=2388) ancient laptop running Lubuntu 18.04 and python 3.7.15 and it clocked 80% (first post demo code, i.e. .0000001 sleep_tune).

Wondering if it was a *buntu thing, I ran Linux MX 19.4 (64bit) off a USB stick (Python 3.7.3) on the same passmark 582 computer: 75% CPU (ran your test code instead of mine, just in case there was some crazy subtle difference.)

So I'm getting slow results on three computers and six different operating system versions, and you're getting fast results on at least two computers and two operating systems... I guess I'm out of ideas... it doesn't seem like it could relate to the board I'm using because it's not even sending/receiving data (I confirmed that with debug print statements.) There's no way I'm ignorant of that the serial connection itself could be affecting things under the hood somehow?

I tried to get a pyserial_asyncio version of telemetrix.py going to see if it might help, but apparently it's a bit beyond my abilities for now -- thought I had it, but in the end couldn't get it to read anything (don't understand asyncio, really.)

MrYsLab, 2022-11-28:
I've [created a file ](https://gist.github.com/MrYsLab/b88d6fbe0cd42f91af4704908831072c)that interacts with the Arduino GPIO pins. It sets up a digital output pin, a PWM output (analog output), and an analog input pin. I have LEDs connected to the digital and PWM outputs and a potentiometer to the analog input.

When I twirl the potentiometer continuously, which sets the brightness of the PWM LED, there is no change in CPU utilization, and the brightness changes are very smooth.

If I change the sleep_tune to 0.1 and run the same script, again twirling the potentiometer does not affect CPU utilization, but this time brightness sometimes stutters. In this case, CPU utilization is 1%.

I ran these tests on a Raspberry Pi 4. Below is a screenshot while running the script with the default sleep_tune; as you can see, it is 16%.

Again, I am unsure why your results are so different from mine.

I am going to close this issue. If you have any additional questions, you may post them here, and I will see them.

[image]

closed 2022-11-28 by MrYsLab

chconnor, 2022-11-28:
I'm ashamed to say that I figured it out. I'm using top, and was misinterpreting the %CPU as overall, rather than per core. I actually checked that before posting the first time (!), but whatever method I used to check it I screwed up, because I drew the conclusion that top was reporting overall CPU and continued with that assumption through all of this.

Very sorry for the wasted time, and thanks for checking things on the Pi as well.

MrYsLab, 2022-11-28:
No need to be ashamed. Mistakes are how we learn, and by the number I have made, I should be a genius by now, but alas, I am not ;-)  I am just glad you figured it out.