MrYsLab/telemetrix #57: multiple servo connections error

Hipposhark, 2023-04-15:
Hello. I have another issue specifically with connecting multiple servos with the `set_pin_mode_servo()` function. In my project, I have connected 5 servos with this function, and they work as expected; however, when I connect a 6th servo, I run the code and there is an exception that is thrown concerning a key error with dht sensors that I do not understand.

[image]

I am not using any dht sensors in my project, so I am not sure what is wrong.
Here is the snippet of code where I initialize the servos.

[image]

If I remove any one of the ArmServos, the code works perfectly, but when I have 6 of them, the error is thrown and the servos do not run. Is there anything that I am doing wrong?

MrYsLab, 2023-04-15:
@Hipposhark 
The error appears due to whatever operations occur when "writing" appears on the screen.

Without seeing the rest of your code, I tried to recreate the issue by running the following code:
```
import sys
import time

from telemetrix import telemetrix

# Create a Telemetrix instance.
board = telemetrix.Telemetrix()
board.set_pin_mode_servo(2, 100, 3000)
time.sleep(.2)
board.set_pin_mode_servo(3, 100, 3000)
time.sleep(.2)
board.set_pin_mode_servo(4, 100, 3000)
time.sleep(.2)
board.set_pin_mode_servo(5, 100, 3000)
time.sleep(.2)
board.set_pin_mode_servo(6, 100, 3000)
time.sleep(.2)
board.set_pin_mode_servo(7, 100, 3000)
for x in range(2, 8):
    board.servo_write(x, 33)
    time.sleep(.3)
time.sleep(3)

board.shutdown()

```
I am unable to reproduce what you are seeing. Please run the code above and let me know if you still get an exception. Also, please provide the rest of your code. If you don't want to post your code here, you can send it to me in an email at the address listed under my name on [this page](https://github.com/MrYsLab).

Thanks.

Hipposhark, 2023-04-15:
Hello. Yes, your code runs without an exception. Here is the rest of the function for you to see. 

```
class ArmServo():
    def __init__ (self, pin, pos=90):
        board.set_pin_mode_servo(pin)
        board.servo_write(pin, pos)

        self.pin = pin
        self.pos = pos

    def read(self):
        return self.pos

    def write(self, new_pos, servo_delay=0.0025):
        print("writing")
        curr_pos = self.pos
        if new_pos == curr_pos:
            return
        elif new_pos < curr_pos:
            for pos in range(curr_pos, new_pos-1, -1):
                board.servo_write(self.pin, pos)
                sleep(servo_delay)
                self.pos = pos
        else:
            for pos in range(curr_pos, new_pos+1):
                board.servo_write(self.pin, pos)
                sleep(servo_delay)
                self.pos = pos

claw = ArmServo(2)           # red
wrist_rotate = ArmServo(3)   # orange
wrist_up_down = ArmServo(4)  # yellow
elbow = ArmServo(5)          # green
shoulder = ArmServo(6)       # blue
base = ArmServo(7)           # violet
```

the write function is suppose is the same as the given `servo_write()` function but with an extra parameter for delay.

MrYsLab, 2023-04-16:
@Hipposhark I modified your code slightly to include the instantiation of the Telemetrix class. I then ran the code and still could not reproduce the issue. Besides the servos, does your program use any other devices, especially input devices? What are those devices, and how are you initializing them in telemetrix to support them?

The modified code is below. If you run that code, do you get the exception?

```
import sys
import time

from telemetrix import telemetrix

class ArmServo:
    def __init__ (self, pin, pos=90):
        board.set_pin_mode_servo(pin)
        board.servo_write(pin, pos)

        self.pin = pin
        self.pos = pos

    def read(self):
        return self.pos

    def write(self, new_pos, servo_delay=0.0025):
        print("writing")
        curr_pos = self.pos
        if new_pos == curr_pos:
            return
        elif new_pos < curr_pos:
            for pos in range(curr_pos, new_pos-1, -1):
                board.servo_write(self.pin, pos)
                sleep(servo_delay)
                self.pos = pos
        else:
            for pos in range(curr_pos, new_pos+1):
                board.servo_write(self.pin, pos)
                sleep(servo_delay)
                self.pos = pos

board = telemetrix.Telemetrix()

claw = ArmServo(2)           # red
wrist_rotate = ArmServo(3)   # orange
wrist_up_down = ArmServo(4)  # yellow
elbow = ArmServo(5)          # green
shoulder = ArmServo(6)       # blue
base = ArmServo(7)           # violet
```

MrYsLab, 2023-04-19:
@Hipposhark Have you had a chance to try and find the source of your issue? If yes, and you determined it is a bug in your code, please close this issue. If not, here are some tips that might help isolate the issue's cause. 

Except for "set_pin_mode_servo," comment out all other calls to set_pin_mode_xxx. Then one by one, enable them and test if the exception occurs. If you find that enabling a pin or pins for one specific mode causes the exception, and if the exception stack trace differs from the one above, please post it here. Also, please let me know which pin mode is the cause.

Since the exception occurs due to a report being generated for phantom DHT modules, this may indicate a bug in the Telemetrix4Arduino sketch or just that you found a path that somehow has the sketch think it should report DHT data. DHT data should be only reported if the [dht_index is non-zero](https://github.com/MrYsLab/Telemetrix4Arduino/blob/e9cd61698678305cd8a2a905b528000dd12026e7/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1894). If the memory location for dht_index is getting set to a non-zero value, we need to find the source of that.

Hipposhark, 2023-04-20:
@MrYsLab Yes sorry for the absence, but I haven't been able to get back to testing out your suggestions. I will update you sometime during the end of this week.

Hipposhark, 2023-04-23:
@MrYsLab Hey so I basically think I may have solved the issue. After adding in some time delay between the initialization calls for the servos, the controls worked perfectly.  I changed 
```
claw = ArmServo(2)           # red
wrist_rotate = ArmServo(3)   # orange
wrist_up_down = ArmServo(4)  # yellow
elbow = ArmServo(5)          # green
shoulder = ArmServo(6)       # blue
base = ArmServo(7)           # violet
```

to 

```claw = ArmServo(2)           # red
sleep(0.1)
wrist_rotate = ArmServo(3)   # orange
sleep(0.1)
wrist_up_down = ArmServo(4)  # yellow
sleep(0.1)
elbow = ArmServo(5)          # green
sleep(0.1)
shoulder = ArmServo(6)       # blue
sleep(0.1)
base = ArmServo(7)           # violet
sleep(0.1)
```
I am using an external library called tkinter as a graphical user interface to control the servos, but all it is doing is running function calls such as 
```
def open_claw():
    claw.write(0)
```
so nothing too fancy.

Basically with 5 servos, the initialization works with or without delay, but when there are six servos, there needed to be delay between each `set_pin_mode_servo` call. I still get the exception involving DHTs without the delay.

MrYsLab, 2023-04-24:
It looks like you solved your problem so I am closing this issue. The tkinter library is a blocking library, so if you are not running it in its own thread, that may cause some issues.
Thanks for letting me know.

diegormsouza, 2023-07-09:
Hi!

I'm not sure if this is a bug so I'm not opening a new issue.

I'm controlling two servos in a Pan-Tilt Gimbal, and the code below works fine, except the fact that when there's a transition between one servo to the other, the first command of this transition results in an abrupt movement, like the position of the servo is reseted somehow.

For example, suppose I just turned the board on, and initialized telemetrix. I start to move the x-axis servo and it runs smooth. When I send the first command to move the y-axis servo, it makes and abrupt movement to the edge (e.g. the gimbal will "look" all the way up), and only then, it starts moving smoothly. Then when I want to move the x-axis servo again, the first movement is also abrupt (all the way to the left in the gimbal). 

It seems like whenever I change servos, it's position is reseted somehow. Is it a timing issue!?

Please find the code below:

```
# required modules
import sys
import time
import keyboard
from telemetrix import telemetrix

# create a Telemetrix instance
board = telemetrix.Telemetrix(com_port='COM4')

# some globals
SERVO_1_PIN = 2
SERVO_2_PIN = 3
LASER_PIN = 4

# initialize servos and laser pins
board.set_pin_mode_servo(SERVO_1_PIN, 100, 3000)
board.set_pin_mode_servo(SERVO_2_PIN, 100, 3000)
board.set_pin_mode_digital_output(LASER_PIN)
time.sleep(.2)

angle  = 120     # initial angle
da = 5           # initial speed (degrees per keypress)
laser_state = 0  # initial led state

while True:

    if keyboard.is_pressed('d'):                  # check if d is pressed
        if angle - da > 0:                        # check to make sure new angle will not exceed 0 degrees
            angle = angle - da                    # decrease angle reference
            board.servo_write(SERVO_1_PIN, angle) # set servo position to new angle by calling the function we made earlier  
            time.sleep(0.1)                       # wait a little bit (0.1 seconds) before checking again  
    
    elif keyboard.is_pressed('a'):                # check if a is pressed
        if angle + da < 180:                      # check to make sure new angle will not exceed 180 degrees
            angle = angle + da                    # increase angle reference
            board.servo_write(SERVO_1_PIN, angle) # set servo position to new angle by calling the function we made earlier
            time.sleep(0.1)                       # wait a little bit (0.1 seconds) before checking again
    
    elif keyboard.is_pressed('w'):                # check if w is pressed
        if angle - da > 0:                        # check to make sure new angle will not exceed 0 degrees
            angle = angle - da                    # decrease angle reference
            board.servo_write(SERVO_2_PIN, angle) # set servo position to new angle by calling the function we made earlier    
            time.sleep(0.1)                       # wait a little bit (0.1 seconds) before checking again

    elif keyboard.is_pressed('s'):                # check if s is pressed
        if angle + da < 180:                      # check to make sure new angle will not exceed 180 degrees
            angle = angle + da                    # if new angle is OK, change to it
            board.servo_write(SERVO_2_PIN, angle) # set servo position to new angle by calling the function we made earlier
            time.sleep(0.1)                       # wait a little bit (0.1 seconds) before checking again

    elif keyboard.is_pressed('t'):                # check if t is pressed
        if da + 1 < 180:                          # check to make sure new angle will not exceed 180 degrees
            da = da + 1                           # increase the speed reference
            time.sleep(0.1)                       # wait a little bit (0.1 seconds) before checking again
   
    elif keyboard.is_pressed('g'):                # check if g is pressed
        if da - 1 > 0:                            # check to make sure new angle will not exceed 0 degrees 
            da = da - 1                           # decrease the speed reference
            time.sleep(0.1)                       # wait a little bit (0.1 seconds) before checking again 
    
    elif keyboard.is_pressed('r'):                # check if r is pressed
        angle = 90                                # set reset angle
        da = 5                                    # set reset speed
        board.servo_write(SERVO_1_PIN, angle)     # set servo position to new angle by calling the function we made earlier
        time.sleep(0.1)                           # wait a little bit (0.1 seconds) before checking again

    elif keyboard.is_pressed('space'):            # if space is pressed toggle the laser
        if laser_state == 0:                      # if laser is off
            board.digital_write(LASER_PIN, 1)     # turn laser on
            laser_state = 1                       
        elif laser_state == 1:                    # if laser is on
            board.digital_write(LASER_PIN, 0)     # turn laser off
            laser_state = 0
        time.sleep(0.2)                           # wait a little bit (0.1 seconds) before checking again

    elif keyboard.is_pressed('esc'):              # if esc is pressed, quit script
        break
```

MrYsLab, 2023-07-09:
@diegormsouza I have two questions:
1. Does initializing the position of the servos fix the problem?
```
# initialize servos and laser pins
board.set_pin_mode_servo(SERVO_1_PIN, 100, 3000)
board.set_pin_mode_servo(SERVO_2_PIN, 100, 3000)
#set starting position of each servo
board.servo_write(SERVO_1_PIN, 120)
board.servo_write(SERVO_2_PIN, 120)
```
2. If that does not help, please tell me the sequence of key presses you are using that cause this behavior.

Thanks.

diegormsouza, 2023-07-09:
Hi,

I have initialized the position of the servos and that does not help.

Regarding the key presses, I have recorded a video showing the problem:

https://github.com/MrYsLab/telemetrix/assets/54595784/f7ee3149-818b-4b97-8612-408a0604a0bd

MrYsLab, 2023-07-09:
Thanks. I will try to recreate the issue here and will report back.

MrYsLab, 2023-07-09:
@diegormsouza Ok. I was able to recreate what you are seeing, but I don't think it is telemetrix. The problem (I believe) is that you are using a single "angle" variable for both servos. 

I made changes to your original program below. Please change the PINs to match your setup and let me know if this works.

I tested using a single physical servo as the "w" "s" servo. If my "fix" still does not work, I will assemble a gimble with two physical servos and then debug.
```
# required modules
import sys
import time
import keyboard
from telemetrix import telemetrix

# create a Telemetrix instance
board = telemetrix.Telemetrix()

# some globals
SERVO_1_PIN = 4
SERVO_2_PIN = 5
LASER_PIN = 4

# initialize servos and laser pins
board.set_pin_mode_servo(SERVO_1_PIN, 100, 3000)
board.set_pin_mode_servo(SERVO_2_PIN, 100, 3000)
board.set_pin_mode_digital_output(LASER_PIN)
time.sleep(.2)

angle = 120  # initial angle
angle2 = 120
da = 5  # initial speed (degrees per keypress)
laser_state = 0  # initial led state

while True:

    if keyboard.is_pressed('d'):  # check if d is pressed
        if angle - da > 0:  # check to make sure new angle will not exceed 0 degrees
            angle = angle - da  # decrease angle reference
            print(f'd: {angle}')

            board.servo_write(SERVO_1_PIN,
                              angle)  # set servo position to new angle by calling the function we made earlier
            time.sleep(0.1)  # wait a little bit (0.1 seconds) before checking again

    elif keyboard.is_pressed('a'):  # check if 'a' is pressed
        if angle + da < 180:  # check to make sure new angle will not exceed 180 degrees
            angle = angle + da  # increase angle reference
            print(f'a: {angle}')

            board.servo_write(SERVO_1_PIN,
                              angle)  # set servo position to new angle by calling the function we made earlier
            time.sleep(0.1)  # wait a little bit (0.1 seconds) before checking again

    elif keyboard.is_pressed('w'):  # check if w is pressed
        if angle2 - da > 0:  # check to make sure new angle will not exceed 0 degrees
            angle2 = angle2 - da  # decrease angle reference
            print(f'w: {angle2}')
            board.servo_write(SERVO_2_PIN,
                              angle2)  # set servo position to new angle by calling the function we made earlier
            time.sleep(0.1)  # wait a little bit (0.1 seconds) before checking again

    elif keyboard.is_pressed('s'):  # check if s is pressed
        if angle2 + da < 180:  # check to make sure new angle will not exceed 180 degrees
            angle2 = angle2 + da  # if new angle is OK, change to it
            print(f's: {angle2}')
            board.servo_write(SERVO_2_PIN,
                              angle2)  # set servo position to new angle by calling the function we made earlier
            time.sleep(0.1)  # wait a little bit (0.1 seconds) before checking again

    elif keyboard.is_pressed('t'):  # check if t is pressed
        if da + 1 < 180:  # check to make sure new angle will not exceed 180 degrees
            da = da + 1  # increase the speed reference
            time.sleep(0.1)  # wait a little bit (0.1 seconds) before checking again

    elif keyboard.is_pressed('g'):  # check if g is pressed
        if da - 1 > 0:  # check to make sure new angle will not exceed 0 degrees
            da = da - 1  # decrease the speed reference
            time.sleep(0.1)  # wait a little bit (0.1 seconds) before checking again

    elif keyboard.is_pressed('r'):  # check if r is pressed
        angle = 90  # set reset angle
        da = 5  # set reset speed
        board.servo_write(SERVO_1_PIN,
                          angle)  # set servo position to new angle by calling the function we made earlier
        time.sleep(0.1)  # wait a little bit (0.1 seconds) before checking again

    elif keyboard.is_pressed('space'):  # if space is pressed toggle the laser
        if laser_state == 0:  # if laser is off
            board.digital_write(LASER_PIN, 1)  # turn laser on
            laser_state = 1
        elif laser_state == 1:  # if laser is on
            board.digital_write(LASER_PIN, 0)  # turn laser off
            laser_state = 0
        time.sleep(0.2)  # wait a little bit (0.1 seconds) before checking again

    elif keyboard.is_pressed('esc'):  # if esc is pressed, quit script
        break
```

diegormsouza, 2023-07-09:
This is totally right, creating a second angle variable solved the problem! (video below)

Thanks a lot! Quick question: I have tried to run the same example with ESP8266 and a message said the SERVO commands were disabled (can't remember the exact message now). How can I enable it?

https://github.com/MrYsLab/telemetrix/assets/54595784/58ded3b5-cfbc-4ecd-a697-a8d278280dfa

MrYsLab, 2023-07-09:
Great! Glad that it works.

The ESP-8266 only works in wireless mode. To use it, the installation instructions are [here](https://mryslab.github.io/telemetrix/telemetrix4esp8266/). I am unsure why you see that the message servo is disabled. One thing to check is to ensure this line in the sketch is set to a value of 1.

If you still get this message, please copy and paste it here.

I am reclosing the issue, but you may reply here if necessary.

closed 2023-07-09 by MrYsLab