MrYsLab/telemetrix #59: How do I implement a rotary encoder with Telemetrix?

GauravMall, 2023-06-20:
I'm trying to use a simple rotary encoder with Telematrix. And I'm basically doing it by translating the simple Arduino code into Python. Basically this:
     
    aState = digitalRead(clockwise_pin);

    if (aState != aLastState) {

        if (digitalRead(anti_clockwise_pin) != aState) {
          counter++;
        } else {
          counter--;
       }

       Serial.print("Position: ");
       Serial.println(counter);
    }

but into Python. Problem is Telematix works with async functions that basically update the value every time it changes, so I can't read the value of pin at that time, or at least I don't know how to. So, what should I do?

diegormsouza, 2023-06-22:
Hi,

I have tested the code provided by Alan and it worked! 

Any ideas on how to add the built-in push button functionality?

Here's an example using the "ButtonFever" library (single press, double press and long press): 
https://github.com/mickey9801/ButtonFever

Here an example sketch that worked here:
   
```
#include <BfButtonManager.h>
#include <BfButton.h>

const unsigned int btnPin = 35;
BfButtonManager manager(btnPin, 4);

BfButton btn1(BfButton::ANALOG_BUTTON_ARRAY, 0);
BfButton btn2(BfButton::ANALOG_BUTTON_ARRAY, 1);
BfButton btn3(BfButton::ANALOG_BUTTON_ARRAY, 2);
BfButton btn4(BfButton::ANALOG_BUTTON_ARRAY, 3);

void pressHandler (BfButton *btn, BfButton::press_pattern_t pattern) {
  Serial.print(btn->getID());
  switch (pattern) {
    case BfButton::SINGLE_PRESS:
      Serial.println(" pressed.");
      break;
    case BfButton::DOUBLE_PRESS:
      Serial.println(" double pressed.");
      break;
    case BfButton::LONG_PRESS:
      Serial.println(" long pressed.");
      break;
  }
}

void specialPressHandler (BfButton *btn, BfButton::press_pattern_t pattern) {
  Serial.print(btn->getID());
  Serial.println(" is special!!");
}

void setup() {
  Serial.begin(115200);
  while (!Serial);
  Serial.println();

  manager.setADCResolution(4096);

  btn1.onPress(pressHandler);
  btn1.onDoublePress(pressHandler);
  btn1.onPressFor(pressHandler, 2000);
  manager.addButton(&btn1, 3000, 3150);
  
  btn2.onPress(pressHandler);
  btn2.onPressFor(pressHandler, 1500);
  manager.addButton(&btn2, 2190, 2350);

  // You may change event handler methods in any order!
  btn3.onPressFor(pressHandler, 1000)
      .onPress(pressHandler)
      .onDoublePress(specialPressHandler, 300);
  manager.addButton(&btn3, 1390, 1550);
  
  btn4.onPress(pressHandler);
  manager.addButton(&btn4, 600, 750);

  manager.begin();
}

void loop() {
  manager.loop();
}
```

And below, the simulation using the example code provided by Alan:

[image]

MrYsLab, 2023-06-22:
@GauravMall 
To use a library such as ButtonFever, it would need to be incorporated into telemetrix as an extension. This involves adding code to the client side (telemetrix.py) and the server side(Telemetrix4Arduino). Something similar was done to support DHT temperature/humidity sensors. A general guide to adding an extension can be [viewed here](https://mryslab.github.io/telemetrix/dht/). It describes adding the DHT extension, but the information is general enough to act as a guide.

I would be happy to provide guidance. However, you would need to implement the extension yourself by forking and modifying both telemetrix and Telemetrix4Arduino.

diegormsouza, 2023-06-22:
@MrYsLab 

I understand. Would the code below be the right approach for the built-in push button functionality?

A have just added another callback and the digital_pin input inside the rotary_simulator def.

It works well! I just wonder if there's a better approach (first time dealing with multiple callbacks).

I imagine that If check the time difference between two presses (with the time library) I can determine if it is a single or a double press (no idea how to determine if it is a long press...).

[image]

```
import sys
import time
from telemetrix import telemetrix

"""
Simulate a rotary encoder
"""

# clock wise, anti clockwise and sw pins
SW_PIN = 11
CLOCK_WISE_PIN = 12 
ANIT_CLOCK_WISE_PIN = 13

# some globals
astate = None
last_state = None
update = False
counter = 0

def the_callback(data):
    """
    A callback function to report data changes

    :param data: [report type, pin, current reported value, pin_mode, timestamp]
    """

    global astate, last_state, update, counter
    the_pin = data[1]
    the_pin_state = data[2]

    if the_pin == CLOCK_WISE_PIN:
        astate = the_pin_state
        if astate != last_state:
            last_state = astate
            update = True
    else:
        if update:
            if the_pin_state != astate:
                counter = counter + 1
            else:
                counter = counter - 1
            update = False

    print(f'astate: {astate}   last_state: {last_state}   update: {update}    counter: '
          f'{counter}')

def the_callback_sw(data):
    sw = data[2]
    if sw == 0:
        print("Button pressed!")

def rotary_simulator(my_board, clockwise_pin, anti_clockwise_pin, sw_pin):
    """
     This function establishes the pins as
     digital inputs. Any changes on the pins will
     be reported through the call back function.

     :param my_board: a telemetrix instance
     :param clockwise_pin: Arduino pin number
     :param anti_clockwise_pin : Arduino pin number
     """

    # set the pin mode
    my_board.set_pin_mode_digital_input(clockwise_pin, the_callback)
    my_board.set_pin_mode_digital_input(anti_clockwise_pin, the_callback)
    my_board.set_pin_mode_digital_input(sw_pin, the_callback_sw)

    try:
        while True:
            time.sleep(.0001)
    except KeyboardInterrupt:
        board.shutdown()
        sys.exit(0)

board = telemetrix.Telemetrix(com_port='COM1')

try:
    rotary_simulator(board, CLOCK_WISE_PIN, ANIT_CLOCK_WISE_PIN, SW_PIN)

except KeyboardInterrupt:
    board.shutdown()
    sys.exit(0)
´´´

diegormsouza, 2023-06-23:
Added the debounce, or a single press could generate more presses:

```
import sys
import time
import datetime
from telemetrix import telemetrix

"""
Simulate a rotary encoder
"""

# clock wise, anti clockwise and sw pins
SW_PIN = 11
CLOCK_WISE_PIN = 12 
ANIT_CLOCK_WISE_PIN = 13

# some globals
astate = None
last_state = None
update = False
counter = 0

# initialize the sw reference datetime
date_previous = datetime.datetime.now()

def the_callback(data):
    """
    A callback function to report data changes

    :param data: [report type, pin, current reported value, pin_mode, timestamp]
    """

    global astate, last_state, update, counter
    the_pin = data[1]
    the_pin_state = data[2]

    if the_pin == CLOCK_WISE_PIN:
        astate = the_pin_state
        if astate != last_state:
            last_state = astate
            update = True
    else:
        if update:
            if the_pin_state != astate:
                counter = counter + 1
            else:
                counter = counter - 1
            update = False

    print(f'astate: {astate}   last_state: {last_state}   update: {update}    counter: '
          f'{counter}')

def the_callback_sw(data):

    global date_previous
    debounceDelay = 0.20

    # get the sw value
    sw = data[2]

    # if sw is equal to zero, the built-in push button has been pressed
    if sw == 0:
        # date time when the sw has been pressed 
        date_press = datetime.datetime.now()
        
        # calculate the difference (absolute) since the last press
        dt = float(abs(date_press - date_previous).total_seconds()) 
        
        # if dt is higher than the debounce time
        if dt > debounceDelay:
                print("Pressed.")
                # update the reference date
                date_previous = datetime.datetime.now()
                
        
def rotary_simulator(my_board, clockwise_pin, anti_clockwise_pin, sw_pin):
    """
     This function establishes the pins as
     digital inputs. Any changes on the pins will
     be reported through the call back function.

     :param my_board: a telemetrix instance
     :param clockwise_pin: Arduino pin number
     :param anti_clockwise_pin : Arduino pin number
     """

    # set the pin mode
    my_board.set_pin_mode_digital_input(clockwise_pin, the_callback)
    my_board.set_pin_mode_digital_input(anti_clockwise_pin, the_callback)
    my_board.set_pin_mode_digital_input_pullup(sw_pin, the_callback_sw)

    try:
        while True:
            time.sleep(.0001)
    except KeyboardInterrupt:
        board.shutdown()
        sys.exit(0)

board = telemetrix.Telemetrix(com_port='COM4')

try:
    rotary_simulator(board, CLOCK_WISE_PIN, ANIT_CLOCK_WISE_PIN, SW_PIN)

except KeyboardInterrupt:
    board.shutdown()
    sys.exit(0)
```

GauravMall, 2023-06-23:
The code does work. I tested it, sorry for the late response, wasn't free. Haven't tested it thoroughly to know if there are hidden bugs, probably not because it's just a rotary encoder implementation. Honestly, I'd recommend an idea for future development, if it's possible and if you want to, to add a simple rotary encoder object. But yeah, it's pretty good, works nicely. Thanks.

MrYsLab, 2023-06-23:
@diegormsouza 

It looks like you've got your problem solved. Nice job!

@GauravMall 
I supported rotary encoders for the [Pymata](https://github.com/MrYsLab/PyMata) project. However, I dropped support because the Arduino encoder libraries supported a small number of processor types. If you like, you can use Pymata and FirmataPlus as a guide to incorporate direct rotary encoder support into telemetrix. I do not plan to add this support in the official releases.

I am going to close this issue. If you have any more comments, feel free to place them here, and I will see them.

closed 2023-06-23 by MrYsLab