MrYsLab/telemetrix #60: Examples not working: Stepper (counterclockwise direction) and i2c LCD

diegormsouza, 2023-06-21:
Hi MrYsLab,

First of all, thank you very much for this great project. Amazing! 

I am trying to reproduce every example from the "examples" folder, using first a simulator and then the real hardware.

For the simulator, I'm using SimulIDE, an open source option (I'm using Arduino UNO in this first test phase).

For now, I have successfully tested in both simulator and real circuit the following examples:
- analog_input.py
- blink.py
- dht.py
- digital_input.py
- fade.py
- hc-sr04_distance_sensor.py
- servo.py
- DC_motor&servo_test_class.py

And the example below using real hardware only (the simulator doesn't have this component):
- i2c_adxl345_accelerometer.py

What gets my attention is that the following examples are not working on both simulator AND the real circuit:
- all the "stepper*" examples are working only for the clockwise direction. When inserting a negative value (counterclockwise), the motor turns in the same direction of the positive value (in the simulator and real circuit). 
- The i2c_lcd_pcf8574_hello_world.py example is also not working in the simulator and real circuit.

Just to confirm my connections are right, I have tested both situations (i2c LCD and stepper) using arduino sketches, and both work well.

So my question is: were you able to test these two examples in a real circuit and it worked?

Thanks a lot!

diegormsouza, 2023-06-21:
Hi!

I forgot to say I'm using a 28BYJ-48 Stepper motor with ULN2003

```
# if you are using a 28BYJ-48 Stepper Motor with ULN2003
# comment out the line above and uncomment out the line below.
motor = the_board.set_pin_mode_stepper(interface=4, pin1=2, pin2=3, pin3=4, pin4=5)
```

MrYsLab, 2023-06-21:
Thanks for letting me know about this. The interface type should equal 8 for the stepper, not 4. [This article](https://community.element14.com/members-area/personalblogs/b/blog/posts/arduino-controlling-stepper-motor-28byj-48-with-accelstepper-library) provides an Arduino sketch. Can you change the example code and test it? If that does not work, could you try the sketch in the article? Please let me know the results. If that change works, I will update the example.

Are any Python console errors printed when you run the LCD example? If yes, could you please provide those? Also, please ensure that the contrast potentiometer on the back of the LCD is adjusted properly. If you adjust the potentiometer, does that solve the problem?

Thanks.

diegormsouza, 2023-06-21:
Thanks!

I will change the interface to 8 and test it again (and try the sketch also). I'll let you know.

As for the LCD, no errors when I run the example, it just doesn't print the message / turn on and off the backlight.
To guarantee I had the potentiometer right (and that the I2C address 0x27 is OK), I have tested the following sketch, and it worked well:

```
//Program: 16x2 LCD display and I2C module

#include <Wire.h>
#include <LiquidCrystal_I2C.h>

// Initialize the display at address 0x27
LiquidCrystal_I2C lcd (0x27,16,2) ; 
 
void setup () {
lcd.init();
}

void loop () {
  lcd.setBacklight(HIGH);
  lcd.setCursor(0,0);
  lcd.print("Hello World!");
  lcd.setCursor(0,1) ;
  lcd.print("LCD and I2C module") ;
  delay(1000);
  lcd.setBacklight (LOW) ;
  delay(1000) ;
}
```

diegormsouza, 2023-06-21:
Hi,

The sketch from the article worked well (1/4 revolution), however when running the Telemetrix example with the same parameters, the stepper is static (it is receiving commands [shaking a little bit] but it doesn't turn).

I have used the same parameters from the sketch, please find below:

`motor = the_board.set_pin_mode_stepper(interface=8, pin1=8, pin2=9, pin3=10, pin4=11)`

```
    # set the max speed and acceleration
    the_board.stepper_set_current_position(0, 0)
    the_board.stepper_set_max_speed(motor, 1000)
    the_board.stepper_set_acceleration(motor, 100)

    # set the absolute position in steps
    the_board.stepper_move_to(motor, 1024)
```

```
    # set the max speed and acceleration
    the_board.stepper_set_current_position(0, 0)
    the_board.stepper_set_max_speed(motor, 1000)
    the_board.stepper_set_acceleration(motor, 100)

    # set the absolute position in steps
    the_board.stepper_move_to(motor, -1024)
```

MrYsLab, 2023-06-21:
Thanks. I will retry the LCD and motor. However, it may take a few days to get to it.

MrYsLab, 2023-06-21:
I just ran the i2c_lcd_pcf8574_hello_world.py script, and it is working for me. Here is [a link](https://www.amazon.com/gp/product/B08LYHC7CQ/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1) to the unit I am using. Since your Arduino Sketch works, I cannot understand why the Python code does not work for you.

I will see if I can get to the motor tomorrow. Could you let me know if you are powering from the Arduino USB or using an external power supply?

diegormsouza, 2023-06-21:
I'm using this same module, but with a protoboard. In the image below I'm using a sketch.
I am powering from the Arduino USB.  Let me test with Telemetrix again. I'll let you know.

[image]

MrYsLab, 2023-06-21:
Thanks. I, too, am powering the LCD from USB power.

I tried the motor, and I am seeing what you are seeing. It may take a little while to understand what is happening.

diegormsouza, 2023-06-21:
Uow, the LCD example just worked! I have deleted my files and downloaded again from GitHub again.
I'm so sorry, I think I have changed something during the tests that made it unusable.
Please let me know if you have any updates about the stepper. 

Thanks a lot! And here's a wishlist: an example to work with keypads.  =)

[image]

MrYsLab, 2023-06-23:
I am still working on understanding what is going on with the 28BYJ-48 Stepper motor with ULN2003. This may take a while for me to track down. I will post when I know why things are failing for the ULN2003 but running for the TB6600.

MrYsLab, 2023-06-24:
To track the motor issue, I created a [new issue](https://github.com/MrYsLab/telemetrix/issues/61). I am at a loss for what the problem might be. I will instrument the AccelStepper library with debug statements and test on a wireless board such as ESP-8266. This will make debugging much more manageable.

I am closing this issue, and you may track progress as it is made by visiting the new issue.

closed 2023-06-24 by MrYsLab