MrYsLab/telemetrix #61: 28BYJ-48 Stepper motor with ULN2003 not working properly

MrYsLab, 2023-06-24:
The motor vibrates but does not turn using this motor/controller combination. Using a NEMA 17 12-volt motor with a TB6600 controller works correctly.

I tried using an external high-current supply for the ULN2003, which did not solve the problem. 

This will take further investigation to determine the cause.

MrYsLab, 2023-06-24:
[@diegormsouza](https://github.com/diegormsouza)
I have figured out what is going on, but unfortunately, the 28BYJ-48 can't be supported by telemetrix.

The reason is that there is a timing issue. When one runs an AccelStepper sketch (non-telemetrix) and an AccelStepper method such as the distanceToGo method is called, the value is available very quickly.

When using telemetrix, however, there is a delay when sending the command to the Arduino and another delay in the returning reply. The run method, therefore, is not being called quickly enough.

When using a controller such as the TB6600, this disparity in timing is internally masked, and things magically work.

I will remove the reference to the 28BYJ-48 in the stepper examples and add a note that the 28BYJ-48 is not supported.

I am closing this issue, but please feel free to leave any further comments here.

closed 2023-06-24 by MrYsLab

diegormsouza, 2023-06-25:
Thanks for verifying that @MrYsLab !

I understand. I'll test a TB6600 controller when I have the opportunity.

stellaurum, 2025-02-11:
@MrYsLab Wanted to check to confirm whether the 28BYJ-48 is supported or not? This comment chain seems to imply not, however the references are still in place within the examples.

MrYsLab, 2025-02-11:
@stellaurum I understand why you are confused.  Based on the testing with the 28BYJ-48 and ULN2003 Driver I probably should have removed the comments. That being said, I found [this code ](https://github.com/BretStateham/28BYJ-48/blob/master/Arduino/Code/S07_AccelStepperDemo/S07_AccelStepperDemo.ino)that supposedly works as an Arduino sketch with Accelstepper. I will test the sketch, and if it works, I will port it to a Telemetrix client and let you know if I am successful or not.
If I am unable to get it working, I will remove the comments.

stellaurum, 2025-02-11:
@MrYsLab I was actually able to get it working most shortly after leaving that comment. As far as I can tell, the Telemetrix examples do, in fact, work mostly as is, the issues I was experiencing that led to me finding this issue chain were a result of typos on my end. That being said, the examples should probably be updated to the following: `set_pin_mode_stepper(interface=4, pin1=8, pin2=9, pin3=10,
    # pin4=11)`. Anyone using the 28BYJ-48 will likely be using the following layout as it is included with purchases of it:

[image]

MrYsLab, 2025-02-11:
Thanks for letting me know. Since it is working for you, I will not be testing it.