MrYsLab/telemetrix #68: No output when digitally reporting more than 13 pins

ysbrandB, 2023-10-28:
For a project I am using your package and an arduino mega connected to a raspberri pi.
I want to connect a whole bunch of buttons to the arduino, so I am using your set_pin_mode_digital_input_pullup function.
It works awesome with 10 buttons, still works when registering the 12th and everything stops working when I register the 13th button.
I have confirmed which pins I use are not the issue, and with the loop_callback function in python the error I get once I register the 13th is the following: 
```Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.9/dist-packages/telemetrix/telemetrix.py", line 2510, in _reporter
    dispatch_entry(response_data)
  File "/usr/local/lib/python3.9/dist-packages/telemetrix/telemetrix.py", line 2198, in _digital_message
    if self.digital_callbacks[pin]:
KeyError: 1```

I hope u can help me with this issue, as up to now this library has been really wonderfull.

ysbrandB, 2023-10-28:
Here is a script to reproduce the issue:
```from telemetrix import telemetrix
import time
class Main:
    def __init__(self):
        self.board = telemetrix.Telemetrix()
        # init password buttons callback
        for i in range(10, 23):
            self.board.set_pin_mode_digital_input_pullup(i, callback=self.ringBack)

    def ringBack(self, state):
        print(state)

if __name__ == '__main__':
    game = Main()
    while True:
        time.sleep(0.0001)```

Which gives the following output:
Retrieving Telemetrix4Arduino firmware ID...
Telemetrix4Arduino firmware version: 5.3.3
[2, 10, 1, 1698534196.6542454]
[2, 12, 1, 1698534196.6546197]
[2, 13, 1, 1698534196.6548982]
[2, 14, 1, 1698534196.6551645]
[2, 15, 1, 1698534196.6555228]
[2, 16, 1, 1698534196.6557324]
[2, 17, 1, 1698534196.656071]
[2, 18, 1, 1698534196.6562645]
[2, 19, 1, 1698534196.6579325]
[2, 20, 1, 1698534196.6582294]
[2, 21, 1, 1698534196.6585016]

As you can see, pin 22 never even gets printed as high for the first time. 
If you make the range to 22 everything works like a charm.

MrYsLab, 2023-10-29:
Thank you for reporting this and for providing the code to reproduce it.
I can reproduce the problem, and it appears it is in Telemetrix4Arduino.

Please give me a little time to debug, and I will report back when I figure out the cause.

MrYsLab, 2023-10-29:
Ok, I found it. It was not a bug in the Python or Arduino side but a race condition.
I added a small delay after each call to set_pin_mode:
```
import time
from telemetrix import telemetrix

class Main:
    def __init__(self):
        self.board = telemetrix.Telemetrix()
        # init password buttons callback
        for i in range(10, 23):
            self.board.set_pin_mode_digital_input_pullup(i, callback=self.ringBack)
            time.sleep(.02)

    def ringBack(self, state):
        print(state)

if __name__ == '__main__':
    game = Main()
    while True:
        time.sleep(0.0001)

```

Here is the output:
```
/home/afy/PycharmProjects/telemetrix311/venv/bin/python /home/afy/PycharmProjects/telemetrix/examples/reported_bugs/2560_many_digital_inputs.py 
Telemetrix:  Version 1.35

Copyright (c) 2021 Alan Yorinks All Rights Reserved.

Opening all potential serial ports...
	/dev/ttyACM1

Waiting 4 seconds(arduino_wait) for Arduino devices to reset...
Valid Arduino ID Found.
Arduino compatible device found and connected to /dev/ttyACM1
Waiting for Arduino to reset
Reset Complete

Retrieving Telemetrix4Arduino firmware ID...
Telemetrix4Arduino firmware version: 5.3.3
[2, 10, 1, 1698590405.4476068]
[2, 11, 1, 1698590405.4477117]
[2, 12, 1, 1698590405.4477818]
[2, 13, 1, 1698590405.467924]
[2, 14, 1, 1698590405.488412]
[2, 15, 1, 1698590405.5088575]
[2, 16, 1, 1698590405.5293987]
[2, 17, 1, 1698590405.5497386]
[2, 18, 1, 1698590405.5662355]
[2, 19, 1, 1698590405.5866716]
[2, 20, 1, 1698590405.6072168]
[2, 21, 1, 1698590405.6276615]
[2, 22, 1, 1698590405.6481597]
```

MrYsLab, 2023-10-30:
Thank you for letting me know. I will add a comment to the user guide.

closed 2023-10-30 by MrYsLab