MrYsLab/telemetrix #69: _i2c_too_many from MCP9600

PierreJeunesse, 2023-12-04:
Hello,
When executing the code below I have an error raised by _i2c_too_many. 
I adapted the example i2c_adxl345_accelerometer.py to the MCP9600 Thermocouple Amplifier from Seeed.
The error occurs randomly after a time between 2 seconds and 2 minutes. 
How to reduce the amount of bytes received form i2c port?
If my example is interesting (and finally works) you can add it to telemetrix/examples.
Best

The code:

```python
"""
 Copyright (c) 2020 Alan Yorinks All rights reserved.

 This program is free software; you can redistribute it and/or
 modify it under the terms of the GNU AFFERO GENERAL PUBLIC LICENSE
 Version 3 as published by the Free Software Foundation; either
 or (at your option) any later version.
 This library is distributed in the hope that it will be useful,
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 General Public License for more details.

 You should have received a copy of the GNU AFFERO GENERAL PUBLIC LICENSE
 along with this library; if not, write to the Free Software
 Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA
"""

"""
This example sets up and control an MCP9600 i2c Thermocouple.
The structure is the same as MCP9600_basic_demo.ino provided by the manufacturer Seeed.
It will continuously print data the raw xyz data from the device.
"""
import sys
import time

from telemetrix import telemetrix

# the call back function to print the MCP9600 data
def the_callback(data):
    """

    :param data: [Device address, device read register, data]
    :return:
    """

    first_byte = data[5]
    sec_byte = data[6]
    Temp_ = first_byte*16 + sec_byte/16
   

    # Afficher le résultat

    print(data)
    print(Temp_)
    

def mpc9600(my_board):
    # setup mpc9600
    # device address = 96
    my_board.set_pin_mode_i2c() # #define DEFAULT_IIC_ADDR  0X60 (in Seeed_MCO9600.h)

    # err_t sensor_basic_config() {
    #     err_t ret = NO_ERROR;
    #     CHECK_RESULT(ret, sensor.set_filt_coefficients(FILT_MID)); (in Seeed_MCO9600.cpp)
    #           IIC_read_byte(THERM_SENS_CFG_REG_ADDR, &therm_cfg_data) (in Seeed_MCO9600.cpp)
    #               THERM_SENS_CFG_REG_ADDR : 0X5 => 5 (in Seeed_MCO9600.h)
    #               FILT_MID : 4 (in Seeed_MCO9600.h)
    my_board.i2c_write(96, [5, 4])
    time.sleep(1)

    #     CHECK_RESULT(ret, sensor.set_cold_junc_resolution(COLD_JUNC_RESOLUTION_0_25));
    #           same logic as above
    my_board.i2c_write(96, [6, 1<<7])
    time.sleep(1)

    #     CHECK_RESULT(ret, sensor.set_ADC_meas_resolution(ADC_14BIT_RESOLUTION));
    my_board.i2c_write(96, [6, 1<<5])
    time.sleep(1)

    #     CHECK_RESULT(ret, sensor.set_burst_mode_samp(BURST_32_SAMPLE));
    my_board.i2c_write(96, [6, 0<<2])
    time.sleep(1)

    #     CHECK_RESULT(ret, sensor.set_sensor_mode(NORMAL_OPERATION));
    my_board.i2c_write(96, [6, 0])
    time.sleep(1)

    while True:
        # read 1 bytes from the data register
        try:
            my_board.i2c_read(96, 0, 2, the_callback, i2c_port=0, write_register=True)
            time.sleep(0.5)

        except (KeyboardInterrupt, RuntimeError):
            my_board.shutdown()
            sys.exit(0)

board = telemetrix.Telemetrix()
try:
    mpc9600(board)
except KeyboardInterrupt:
    board.shutdown()
    sys.exit(0)
```

The error:

Exception in thread Thread-2 (_reporter):
Traceback (most recent call last):
  File "C:\ProgramData\miniconda3\envs\pymo4_dev\Lib\threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "C:\ProgramData\miniconda3\envs\pymo4_dev\Lib\threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Users\Pierre\Documents\GitHub\telemetrix\telemetrix\telemetrix.py", line 2510, in _reporter
    dispatch_entry(response_data)
  File "C:\Users\Pierre\Documents\GitHub\telemetrix\telemetrix\telemetrix.py", line 2257, in _i2c_too_many
    raise RuntimeError(
RuntimeError: i2c too many bytes received from i2c port 1 i2c address 96
object address  : 000002340AB66620
object refcount : 2
object type     : 00007FFFB8DFEBF0
object type name: RuntimeError
object repr     : RuntimeError('Shutdown failed - could not send stop streaming message')
lost sys.stderr

MrYsLab, 2023-12-04:
Thanks for reporting this. I don't have any experience with the MCP9600. However, looking at the [Seeed Studio basic example](https://github.com/Seeed-Studio/Seeed_MCP9600/blob/master/examples/MCP9600_basic_demo/MCP9600_basic_demo.ino), the read loop has a delay of 1 second. Please try expanding the delay in your code from .5 seconds to 1 second and see if that helps.

PierreJeunesse, 2023-12-05:
Thanks for your fast answer. I set 1 second to all parameter without effect. Thanks to your advice I checked with 2 seconds delay but the error remains.

I'll dig in the i2c rules to understand what "too many bytes" actually means:
-too much at a time?=> what is the threshold?
-more than expected?=> How to set what is expected?
-too much in memory?=> How to empty memory after each read?
-etc...

MrYsLab, 2023-12-05:
Thanks for letting me know. I hoped that the fix might be that simple. "Too many bytes" means that the i2c read returned more than the requested number of bytes. This is defined in the[ i2c_read](https://github.com/MrYsLab/Telemetrix4Arduino/blob/6883fcd978cfb5a31882a6f0b335f099df1e8d1a/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L969) function of Telemetrix4Arduino. This is detected [here.](https://github.com/MrYsLab/Telemetrix4Arduino/blob/6883fcd978cfb5a31882a6f0b335f099df1e8d1a/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1017)

Please be aware that the comment at the beginning of the function needs to be corrected. It should be:
```
// data in the incoming message:

  // address, [0]
  // register, [1]
  // number of bytes, [2]
  // stop transmitting flag [3]
  // i2c port 4]
  // write the register [5]
```
I will fix it in the next release.

Most likely, you may have missed initializing a register in the device, causing the issue.

Porting Arduino i2c code is sometimes rather complex, depending on the device. Verify that your Python code is initializing and setting the registers precisely as is done in the Arduino library supporting the device. You don't have to recreate all the methods of the Arduino library, but you must handle the registers as the library does.

MrYsLab, 2023-12-05:
I forgot to include the call to [_**requestFrom**_ ](https://github.com/arduino/ArduinoCore-avr/blob/63092126a406402022f943ac048fa195ed7e944b/libraries/Wire/src/Wire.cpp#L173) in the Arduino Wire Library.

MrYsLab, 2023-12-06:
I am closing this issue because I don't own the i2c device in question and will not be able to reproduce the error.
Please feel free to post any questions here; I will be pleased to answer.
If need be, I can always reopen the issue.

closed 2023-12-06 by MrYsLab

MrYsLab, 2023-12-07:
Great job in figuring this out! 
I have added your file to the [contributed examples](https://github.com/MrYsLab/telemetrix/blob/master/examples/Contributed/i2c_MCP9600_thermocouple.py). Also, I changed the copyright statement. I hope that is ok.

Release 1.36 contains the change.

Thanks again.

MrYsLab, 2023-12-07:
:+1:  :smile: