MrYsLab/telemetrix #73: Mock interface for unit testing

jacopoabramo, 2024-02-23:
I'm currently refactoring a Telemetrix interface for a stepper motor controlled via an Arduino in collaboration with another developer. I do not have access to their hardware because we live in separate locations and are interacting mostly online. For our application I would like to implement a mock interface that replicates Telemetrix callback system. Is there something readily available or should I do that from scratch? At this time I'm following the second option (we're using a limited subset of the available APIs) but I'm struggling on how to replicate the callback behavior.

MrYsLab, 2024-02-23:
@jacopoabramo If you have access to an Arduino, the simplest thing would be to connect a pushbutton switch to a digital input. You can then treat the callback for the switch closure as if it came from a stepper completing its cycle. The stepper callback data is straightforward since it only identifies the motor ID sending the callback. The id is located in data[1] of the callback. For the digital input callback, the data located in data[1] is the pin number. You can treat that value as the motor ID. This way, you will exercise the callback functionality through the normal threaded channels.

If you did not have access to any hardware at all, then you might wish to use something like [APScheduler](https://apscheduler.readthedocs.io/en/3.x/index.html) to call the callback method in your code.

I hope that helps answer your question. If it does, please close this issue. If you need help, please let me know.

jacopoabramo, 2024-02-23:
I will check the package you mentioned. Thank you, closing.

closed 2024-02-23 by jacopoabramo