MrYsLab/telemetrix #77: Creating a second callback for a DHT sensor causes it report errors

nariox, 2024-03-20:
Hello,

I was using telemetrix to develop an application, when I stumbled upon an issue. If I call `set_pin_mode_dht` twice on the same pin causes the board to report twice per readout, but the second report comes up with code 254 (`DHTLIB_ERROR_TIMEOUT` from the `DHTstable` library). The reported data is, however, still valid (If I modify `telemetrix.py` to ignore `data[0]` and convert the packet regardless I get repeat readings).

I have a few ideas to mitigate that:

1. Add a check for whether `self.dht_callbacks[pin]` exists, if so, trigger an error.
2. Turn `dht_callbacks` into a dictionary of lists. Then multiple callbacks can be set for a single sensor.
3. Add a check for whether `self.dht_callbacks[pin]` exists, if so, only replace the old callback with the new callback, but don't send `DHT_NEW` to the board (and don't increment `dht_count`).

I'm willing to write a PR for this (I already have a internal "workaround" using idea 3), but I would rather not use an option that won't be upstreamed. Idea 1 would also work for my purposes (as long as it does not trigger `self.shutdown()`). Idea 2 would probably be somewhat troublesome to write and perhaps of little usefulness (with the exception of exceedingly lazy developers like me).

Let me know what you think
Thanks

MrYsLab, 2024-03-20:
Thanks for reporting this. Item 1 is the simplest to implement. I have a question on item 3. Why would one need to change the callback after initializing a callback on a pin?
I am leaning towards item one, where if there already is a callback assigned to the pin, raise a runtime error.

nariox, 2024-03-20:
tl;dr: I think item 1 makes a lot of sense, I'd be fine with it.

medium length answer: I have a program where one could dynamically assign "sensors" upon initialization*, but they only allow one variable to be assigned to an object (for example, either temperature or humidity). Each "sensor" is initialized with a line such as "telemetrix instance_id method pin_number" (for example "telemetrix 1 analog_in 0"). In the inialization for each of these sensors, I had a corresponding `set_pin_mode...` function. This works fine for things like analog/digital in/out, but DHT presented an issue, because once it would be `set_pin_mode_dht` ran twice for the a single pin (triggering my issues).

My original idea was to set a callback to update temperature and then one to update the humidity. This obviously didn't work. So I created a structure that buffers the data from the DHT and triggers the callbacks for both "sensor" objects.

Reassigning the callback saves me from having to keep track of which pins/callbacks I have already set up. Of course, item 1 works just as well, as long as I envelop the `set_pin_mode_dht` in a `try/except` or set up proper bookkeeping. So I think option is the more sane option (to avoid unpredictable behavior and force better programming practices). 

* the application is called an Input/Output Controller (IOC) and is part of a framework called EPICS, which is used in soft real-time distributed control networks. I don't that's too important for this conversation, but thought you might want to know.

MrYsLab, 2024-03-20:
To make sure, you would like the change to be made for this repository, telemetrix. I will also implement telemetrix-aio.

Please verify so that I can modify the correct repositories.

MrYsLab, 2024-03-21:
I have an updated version of telemetrix.py in [this branch.](https://github.com/MrYsLab/telemetrix/tree/dht_pin_dup). Please let me know if this solves your issue, and I will create a new release.

nariox, 2024-03-26:
Sorry, I had to travel out of town last week so I could only get back to you now. I had basically the same commit here: https://github.com/MrYsLab/telemetrix/compare/master...nariox:telemetrix:warndoublecallback but yours works.

Thank you again.

MrYsLab, 2024-03-26:
Thanks. Version 1.39 with the fix is now available on PyPi .

closed 2024-03-26 by MrYsLab