MrYsLab/telemetrix #90: Multiple Arduinos

noerK, 2024-11-06:
Hey!

First of all thank you for the all the work on these libs. The last days i started digging into firmata and ended up here.

### My Setup:
1 x Raspberry Pi 4
3 x Arduino Nano

Basically this is a proof of concept to control "modules" for a project.

I followed the [documentation](https://mryslab.github.io/telemetrix/telemetrix4arduino/) and modified the sketch for each arduino accordingly (`#define ARDUINO_ID $id`)

Then i tried creating multiple instances like this:
`board1 = telemetrix.Telemetrix(arduino_instance_id=1, arduino_wait=4)` (same for board 2 and 3)

### Whats working:

- all 3 boards are detected 
- if not multiple instances are created every board is working solo
- if there is only 1 instance loopback is working

### Issues:

If i am trying to setup multiple instances via ID there are several errors popping up:

`IndexError: list index out of range`
or
`TypeError: 'NoneType' object is not callable`

They occur kinda random - thus i updated the script a bit in order to debug better and create a proper setup with multiple attempts. I also played around with the arduino_wait time. The setup is working after some failed attempts BUT the actual functionality is not working - not all loopbacks are arriving.

### Script:

`import time
import sys
from telemetrix import telemetrix

"""
Loopback some data to assure that data can be sent and received between
the Telemetrix client and arduino-telemetrix server.
"""

def the_callback(data):
    """
    A callback function to report receive the looped back data

    :param data: [looped back data]
    """
    print(f'Looped back: {chr(data[0])}')

def loop_back(my_board, loop_back_data):
    """
    This function will request that the supplied characters be
    sent to the board and looped back and printed out to the console.

    :param my_board: a telemetrix instance
    :param loop_back_data: A list of characters to have looped back
    """
    try:
        for data in loop_back_data:
            my_board.loop_back(data, callback=the_callback)
            print(f'Sending: {data}')
    except KeyboardInterrupt:
        board1.shutdown()
        board2.shutdown()
        board3.shutdown()
        sys.exit(0)

board1 = None
board2 = None
board3 = None

def setup_boards(skip_1: False, skip_2: False, skip_3: False):
    global board1, board2, board3
    max_attempts = 10
    attempt = 0
    print('setup')
    while(attempt < max_attempts):
        if board1 is None and skip_1 == False:
            print('trying to setup board 1')
            try:
                board1 = telemetrix.Telemetrix(arduino_instance_id=1, arduino_wait=4)
            except:
                print('error board 1')
        else:
            print('board 1 setup complete')

        if board2 is None and skip_2 == False:
            print('trying to setup board 2')
            try:
                board2 = telemetrix.Telemetrix(arduino_instance_id=2, arduino_wait=4)
            except:
                print('error board 2')
        else:
            print('board 2 setup complete')

        if board3 is None and skip_3 == False:
            print('trying to setup board 3')
            try:
                board3 = telemetrix.Telemetrix(arduino_instance_id=3, arduino_wait=4)
            except:
                print('error board 3')
        else:
            print('board 3 setup complete')

        if (board1 is not None or skip_1 == True) and (board2 is not None or skip_2 == True) and (board3 is not None or skip_3 == True):
            print('setup complete')
            break

        attempt += 1
        time.sleep(2)

char_list1 = ['A', 'B', 'C']
char_list2 = ['D', 'E', 'F']
char_list3 = ['G', 'H', 'I']

setup_boards(True, False, True)

try:
    if board1 is not None:
        loop_back(board1, char_list1)
    if board2 is not None:
        loop_back(board2, char_list2)
    if board3 is not None:
        loop_back(board3, char_list3)
    time.sleep(5)
except KeyboardInterrupt:
    board1.shutdown()
    board2.shutdown()
    board3.shutdown()
    sys.exit(0)
`

### Versions:

python3.11
Telemetrix:  Version 1.41
Telemetrix4Arduino firmware version: 5.4.4

Thanks for any help in advance.
Best regards

noerK, 2024-11-06:
Update:

I just tinkered around a bit and found out that everything is working fine if i add the correct com_port to each board. I updated the setup method to loop through USB Ports. It feels that if there are too many attempts of creating an instance it causes issues:
`Exception in thread Thread-5 (_reporter):
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/charly2/my-venv/lib/python3.11/site-packages/telemetrix/telemetrix.py", line 2523, in _reporter
    dispatch_entry(response_data)
  File "/home/charly2/my-venv/lib/python3.11/site-packages/telemetrix/telemetrix.py", line 2208, in _digital_message
    value = data[1]
            ~~~~^^^
IndexError: list index out of range`

MrYsLab, 2024-11-06:
Thanks for reporting this and providing a detailed description. I can confirm the issue, and you are correct that if the com_port is specified, the problem does not exhibit itself.

It may take me a few days to debug this issue, and I will post once I understand its nature.

noerK, 2024-11-06:
Wow  - thats a fast response! Thanks for looking into this. If you need any more information let me know and i'll provide it.

MrYsLab, 2024-11-06:
I can only get multiple boards working consistently by specifying the com_port and arduino_id when instantiating telemetrix. 

The latest code release has a small code change that I thought fixed the problem, but with that change, things occasionally fail unless the arduino_id and com_port are specified.

I updated the [User's Guide](https://mryslab.github.io/telemetrix/telemetrix4arduino/#setting-the-telemetrix4arduino-instance-id) stating that the two parameters must be specified when instantiating telemetrix.

I am going to close the issue. If you wish to add a comment you may do so here, and I will set it.

Thanks again.

closed 2024-11-06 by MrYsLab

MrYsLab, 2024-11-06:
I decided to try running a program without specifying the com_port, and it is now working.

Here is what I ran:
```
import time
import sys
from telemetrix import telemetrix

"""
Loopback some data to assure that data can be sent and received between
the Telemetrix client and arduino-telemetrix server.
"""

def the_callback(data):
    """
    A callback function to report receive the looped back data

    :param data: [looped back data]
    """
    print(f'Looped back: {chr(data[0])}')

def loop_back(my_board, loop_back_data):
    """
    This function will request that the supplied characters be
    sent to the board and looped back and printed out to the console.

    :param my_board: a telemetrix instance
    :param loop_back_data: A list of characters to have looped back
    """
    try:
        for data in loop_back_data:
            my_board.loop_back(data, callback=the_callback)
            print(f'Sending: {data}')
    except KeyboardInterrupt:
        board.shutdown()
        sys.exit(0)

# board = telemetrix.Telemetrix(com_port='/dev/ttyACM0', arduino_instance_id=1)
# board2 = telemetrix.Telemetrix(com_port='/dev/ttyACM1', arduino_instance_id=2)

board = telemetrix.Telemetrix(arduino_instance_id=1)
board2 = telemetrix.Telemetrix(arduino_instance_id=2)

char_list = ['A', 'B', 'C']
char_list2 = ['X', 'Y', 'Z']

try:
    while True:
        loop_back(board, char_list)
        loop_back(board2, char_list2)
except KeyboardInterrupt:
    board.shutdown()
    sys.exit(0)

```
Could you please try this program and let me know if it works? I tried with both version 1.41 and 1.42 and it works with both.

noerK, 2024-11-07:
Hey i also digged a bit into the source code yesterday and i **think** i got to root of the issue.

After your response yesterday i tried to run it with the new condition you introduced but also changed the initial value of `reported_arduino_id` to `None`. This worked for 2 devices... as soon as i connected the third one it didn't anymore. Then i thought i might have something to do with the order.

If i didn't miss anything each new instance always tries to create a `Serial` connection for each port. Also if the right board was found, all other connections are kept open because `close()` is never called, right? I never worked with `Serial` in beforehand but i assumed this could cause trouble. 

So i came up with a little change to tell instance 2 and 3 which ports to avoid without introducing breaking changes or make the "normal" setup more complicated. I introduced a new param called `excluded_ports` which can be added to the init:
```
board1 = telemetrix.Telemetrix(arduino_instance_id=1, arduino_wait=2)
board2 = telemetrix.Telemetrix(arduino_instance_id=2, arduino_wait=2, excluded_ports=[board1.serial_port])
board3 = telemetrix.Telemetrix(arduino_instance_id=3, arduino_wait=2, excluded_ports=[board1.serial_port, board2.serial_port])
```

Make sure connections are closed if the first ones in the loop are not matching:
```
if self.reported_arduino_id != self.arduino_instance_id:
    self.serial_port.close()
    continue
```

Ignore all ports from the `excluded_ports` list:

```
for port in the_ports_list:
    if port.pid is None or port.device in port_filter:
        continue
    try:
        self.serial_port = serial.Serial(port.device, 115200, timeout=1, writeTimeout=0)
    except SerialException:
        continue
```

In my tests yesterday it worked (but it was also late :D).
I will give your script from above a try and also if i can break it with a third one.
If i validated everything i can open a PR if needed & you want to.

noerK, 2024-11-07:
UPDATE:
I tested the code above with 3 devices connected:
```
Traceback (most recent call last):
  File "/usr/lib/python3.11/threading.py", line 1038, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.11/threading.py", line 975, in run
    self._target(*self._args, **self._kwargs)
  File "/home/charly2/my-venv/lib/python3.11/site-packages/telemetrix/telemetrix.py", line 2525, in _reporter
    dispatch_entry(response_data)
  File "/home/charly2/my-venv/lib/python3.11/site-packages/telemetrix/telemetrix.py", line 2210, in _digital_message
    value = data[1]
            ~~~~^^^
IndexError: list index out of range
Traceback (most recent call last):
  File "/home/charly2/telemetrix_test.py", line 41, in <module>
    board2 = telemetrix.Telemetrix(arduino_instance_id=2)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/charly2/my-venv/lib/python3.11/site-packages/telemetrix/telemetrix.py", line 275, in __init__
    self._find_arduino()
  File "/home/charly2/my-venv/lib/python3.11/site-packages/telemetrix/telemetrix.py", line 391, in _find_arduino
    raise RuntimeError(f'Incorrect Arduino ID: {self.reported_arduino_id}')
RuntimeError: Incorrect Arduino ID: []
```

I am preparing my PR - its easier discussing than sending pieces of code in comments :D

MrYsLab, 2024-11-07:
Thanks for all the effort you are putting into this.

noerK, 2024-11-07:
Haha you're welcome :D I am happy contributing.

I am now successfully running 5 Arduinos on my raspberry pi!
[image]

I also had to find out that not all USB hubs have a proper quality level to be used for serial connections on a raspberry it seems. I switched it and now everything works.

#91