MrYsLab/telemetrix #92: Can't read Digital Input at the start

ZdTmasar, 2025-02-26:
Hi,

I agree on the Callback strategy being more efficient than polling.
But when I start my application, and I set a digital pin as input, I don't know what is the value of this pin. And I will never know until its value change.

Maybe there is a way to read the "first" value of the digital pin but I can"t seem to find one.

Could someone help me on this issue please ?

Thank you !

MrYsLab, 2025-02-26:
Thanks for the question.

When the Telemtrix4Arduino sketch [initializes](https://github.com/MrYsLab/Telemetrix4Arduino/blob/4e63d385d2fc87c490f141fb363b79c84fe04c6e/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L2012) for digital inputs, it sets the [last value read to LOW](https://github.com/MrYsLab/Telemetrix4Arduino/blob/4e63d385d2fc87c490f141fb363b79c84fe04c6e/examples/Telemetrix4Arduino/Telemetrix4Arduino.ino#L1722) (independent of the actual state of the pin). If the digital input is HIGH when the sketch initializes, the callback will be sent almost immediately since the state is perceived to have changed.

If no callback is issued at startup, the state of the pin is LOW.

Please let me know if you have any further questions.

ZdTmasar, 2025-02-26:
Thanks for your answer.
I can't seem to reproduce the behavior you described. I have the following python code :

``
from telemetrix import telemetrix
import time
_CB_PIN_MODE = 0
_CB_PIN = 1
_CB_VALUE = 2
_CB_TIME = 3
def the_callback(data) -> None:
    """
        callback update value
        :param data: [pin, current reported value, pin_mode, timestamp]
    """
    date = time.strftime("%Y-%m-%d %H:%M:%S", time.localtime(data[_CB_TIME]))
    print(f"Pin Mode: {data[_CB_PIN_MODE]} Pin: {data[_CB_PIN]} Value: {data[_CB_VALUE]} Time Stamp: {date}")
_my_board = telemetrix.Telemetrix()
_my_board.set_pin_mode_digital_input(13, the_callback)

while True:
    time.sleep(1)
``

On the digital pin 13, I tried with GND linked to it, started the script. Did the same with 3.3V. In both cases, I have no callback at the start of the board. It's a Arduino Leonardo.

I also tries on another digital input.

Do you have an idea on why ?

MrYsLab, 2025-02-26:
I just tried reproducing your example. I am using an Arduino Leonardo. I  connected a jumper between 3.3v on the Arduino and pin 13. Here is the output:
```
Waiting 4 seconds(arduino_wait) for Arduino devices to reset...
Arduino compatible device found and connected to /dev/ttyACM0
Waiting for Arduino to reset
Reset Complete
Retrieving Arduino ID...
Valid Arduino ID Found.

Retrieving Telemetrix4Arduino firmware ID...
Telemetrix4Arduino firmware version: 5.4
Enter Control-C to quit.
Pin Mode: 2 Pin: 13 Value: 1 Time Stamp: 2025-02-26 13:20:51

```

I am not sure why it is not working for you.  Are you sure that the 3.3v hasn't dropped for some reason?

MrYsLab, 2025-02-26:
Here is the test program I am using:

```
"""
 Copyright (c) 2020 Alan Yorinks All rights reserved.

 This program is free software; you can redistribute it and/or
 modify it under the terms of the GNU AFFERO GENERAL PUBLIC LICENSE
 Version 3 as published by the Free Software Foundation; either
 or (at your option) any later version.
 This library is distributed in the hope that it will be useful,f
 but WITHOUT ANY WARRANTY; without even the implied warranty of
 MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
 General Public License for more details.

 You should have received a copy of the GNU AFFERO GENERAL PUBLIC LICENSE
 along with this library; if not, write to the Free Software
 Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA

 DHT support courtesy of Martyn Wheeler
 Based on the DHTNew library - https://github.com/RobTillaart/DHTNew
"""

import sys
import time

from telemetrix import telemetrix

"""
Monitor a digital input pin
"""

"""
Setup a pin for digital input and monitor its changes
"""

# Setup a pin for analog input and monitor its changes
DIGITAL_PIN = 13  # arduino pin number

# Callback data indices
CB_PIN_MODE = 0
CB_PIN = 1
CB_VALUE = 2
CB_TIME = 3

def the_callback(data):
    """
    A callback function to report data changes.
    This will print the pin number, its reported value and
    the date and time when the change occurred

    :param data: [pin, current reported value, pin_mode, timestamp]
    """
    date = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(data[CB_TIME]))
    print(f'Pin Mode: {data[CB_PIN_MODE]} Pin: {data[CB_PIN]} Value: {data[CB_VALUE]} Time Stamp: {date}')

def digital_in(my_board, pin):
    """
     This function establishes the pin as a
     digital input. Any changes on this pin will
     be reported through the call back function.

     :param my_board: a pymata4 instance
     :param pin: Arduino pin number
     """

    # set the pin mode
    my_board.set_pin_mode_digital_input(pin, callback=the_callback)
    # time.sleep(1)
    # my_board.disable_all_reporting()
    # time.sleep(4)
    # my_board.enable_digital_reporting(12)

    # time.sleep(3)
    # my_board.enable_digital_reporting(pin)
    # time.sleep(1)

    print('Enter Control-C to quit.')
    # my_board.enable_digital_reporting(12)
    try:
        while True:
            time.sleep(.0001)
    except KeyboardInterrupt:
        board.shutdown()
        sys.exit(0)

board = telemetrix.Telemetrix()

try:
    digital_in(board, DIGITAL_PIN)
except KeyboardInterrupt:
    board.shutdown()
    sys.exit(0)

```

MrYsLab, 2025-02-28:
I am unsure if you have solved your issue, but since I cannot reproduce it, I am closing the issue. You may leave a comment here and I will see it. If necessary, the issue may be reopened

closed 2025-02-28 by MrYsLab