MrYsLab/telemetrix #93: Unable to control more than 2 stepper motors

SilverSpartan13, 2025-03-19:
I’m currently working on a project that involves controlling four stepper motors using an Arduino Mega, a RAMPS 1.4 shield, and A4988 stepper drivers. So far, I’ve successfully controlled each stepper motor individually by modifying the pin assignments in the provided examples. Additionally, I was able to adapt the examples to alternate control between two stepper motors. However, when I try to add a third motor, I encounter an issue where none of the motors move.

I’m looking for suggestions to debug this problem or an example code that effectively controls multiple stepper motors (three or more) simultaneously. Any guidance would be greatly appreciated.

## Here is the code I've used to test motor movement:
```
import sys
import time
from telemetrix import telemetrix

# Initialize Telemetrix instance
board = telemetrix.Telemetrix()

def the_callback(data):
    date = time.strftime('%Y-%m-%d %H:%M:%S', time.localtime(data[2]))
    print(f'Run motor {data[1]} completed motion at: {date}.')

# Pin configuration
stepper_configs = [
    {'pulse_pin': 54, 'dir_pin': 55, 'enable_pin': 38},  # Stepper 0 Ramps_X
    {'pulse_pin': 60, 'dir_pin': 61, 'enable_pin': 56},  # Stepper 1 Ramps_Y
    {'pulse_pin': 46, 'dir_pin': 48, 'enable_pin': 62},  # Stepper 2 Ramps_Z
    {'pulse_pin': 26, 'dir_pin': 28, 'enable_pin': 24}   # Stepper 3 Ramps_E0
]

steppers = []
# Initialize steppers
for config in stepper_configs:
    # Set up the stepper motor with the specified pins
    motor = board.set_pin_mode_stepper(interface=1, pin1=config['pulse_pin'], pin2=config['dir_pin'])
    
    # Set the enable pin as a digital output
    board.set_pin_mode_digital_output(config['enable_pin'])
    steppers.append(motor)
    
# Set max speed for steppers
for stepper in steppers:
    board.stepper_set_max_speed(stepper, 999)
    board.stepper_set_acceleration(stepper, 999)

# Stepper run
def run_stepper(stepper_num):
    print(f"Running Stepper {stepper_num}.")
    board.stepper_set_speed(steppers[stepper_num], 500) # Set the speed of the motor
    board.digital_write(stepper_configs[stepper_num]['enable_pin'], 0) # Send enable pin low to enable stepper(Active low)
    board.stepper_run_speed(steppers[stepper_num]) # Tell motor to run
    time.sleep(2)
    board.stepper_stop(steppers[stepper_num]) # Stop stepper
    board.digital_write(stepper_configs[stepper_num]['enable_pin'], 1) # Send enable pin high to disable stepper

# Main loop
while True:
    for i in steppers:
        run_stepper(i)

    time.sleep(1)  # Adjust the loop delay as needed
```

MrYsLab, 2025-03-19:
To eliminate telemetrix as a possible cause, could you please modify [this Arduino sketch](https://github.com/waspinator/AccelStepper/blob/master/examples/ConstantSpeed/ConstantSpeed.pde) to support your four motors?

If this fails with more than two motors, then you may have a power supply issue

SilverSpartan13, 2025-03-19:
I was able to move all of the motors with this Arduino code:
```
#include <AccelStepper.h>
#include <loopTimer.h>

#define Num_of_steppers 4
int stepper_configs[Num_of_steppers][3] = {//pulse_pin, dir_pin, enable_pin
    {54, 55, 38},  // Stepper 0 Ramps_X
    {60, 61, 56},  // Stepper 1 Ramps_Y
    {46, 48, 62},  // Stepper 2 Ramps_Z
    {26, 28, 24}   // Stepper 3 Ramps_E0
};
int selectedStepper = 0;

AccelStepper* Steppers[Num_of_steppers];
millisDelay SwitchStepper;

void CheckChangeStepper() {
  if (SwitchStepper.justFinished()) {
    SwitchStepper.repeat(); // start delay again without drift
    digitalWrite(stepper_configs[selectedStepper][2],HIGH);//turn off stepper
    selectedStepper++;
    if(selectedStepper >= Num_of_steppers){
      selectedStepper = 0;
    }
    digitalWrite(stepper_configs[selectedStepper][2],LOW);//turn on new stepper
  } // else nothing to do this call just return, quickly
}

void setup(){  
  for(int i = 0; i < Num_of_steppers; i++){
    Steppers[i] = new AccelStepper(AccelStepper::DRIVER, stepper_configs[i][0], stepper_configs[i][1]);
    Steppers[i] -> setMaxSpeed(1000);
    Steppers[i] -> setSpeed(500);	
    pinMode(stepper_configs[i][2],OUTPUT);
  }
  digitalWrite(stepper_configs[selectedStepper][2],LOW);
  SwitchStepper.start(2000);
}

void loop(){  
  Steppers[selectedStepper] -> runSpeed();
  CheckChangeStepper();
}
```

MrYsLab, 2025-03-20:
Thank you for testing this. To help determine the cause of the issue, I would like to create one or more Telemetirx scripts for you to try. I do not have four motors, or I would test things here.

Do you intend to run the motors in continuous mode for your application, or will you use relative or absolute positioning or some combination of modes?

MrYsLab, 2025-03-20:
Thanks. I will create the scripts and let you know when they are available. It may take me a day or so.

MrYsLab, 2025-03-21:
Please try the following script and let me know if it works:
```
import sys
import time
from telemetrix import telemetrix

# Initialize Telemetrix instance
board = telemetrix.Telemetrix()

# Pin configuration
stepper_configs = [
    {'pulse_pin': 54, 'dir_pin': 55, 'enable_pin': 38},  # Stepper 0 Ramps_X
    {'pulse_pin': 60, 'dir_pin': 61, 'enable_pin': 56},  # Stepper 1 Ramps_Y
    {'pulse_pin': 46, 'dir_pin': 48, 'enable_pin': 62},  # Stepper 2 Ramps_Z
    {'pulse_pin': 26, 'dir_pin': 28, 'enable_pin': 24}  # Stepper 3 Ramps_E0
]

steppers = []
motors = []

# Initialize steppers
for config in stepper_configs:
    # Set up the stepper motor with the specified pins
    motor = board.set_pin_mode_stepper(interface=1, pin1=config['pulse_pin'],
                                       pin2=config['dir_pin'])
    board.stepper_set_max_speed(motor, 999)
    board.stepper_set_speed(motor, 500)
    # Set the enable pin as a digital output
    board.set_pin_mode_digital_output(config['enable_pin'])
    board.digital_write(config['enable_pin'], 0)
    steppers.append(motor)

# Main loop
try:
    while True:
        for motor in motors:
            # run the motors
            board.stepper_run_speed(motor)
            time.sleep(.2)
except KeyboardInterrupt:
    for config in stepper_configs:
        board.digital_write(config['enable_pin'], 1)
    board.shutdown()
    sys.exit(0)
```

SilverSpartan13, 2025-03-22:
I've run your program, and there is no movement on any steppers. Even when commenting out steppers 2 and 3 from the pin config.

MrYsLab, 2025-04-03:
It has been two weeks since my last comment. I am going to close this issue because I have not heard back.

closed 2025-04-03 by MrYsLab